lbr-stack/lbr_fri_ros2_stack #193: Control arm LBR only on RVIZ without gazebo

antonio1matos, 2024-06-26:
Hello. 
I was using this code: ros2 launch lbr_bringup bringup.launch.py model:=iiwa14 moveit:=true to launch the robot iiwa14 in simulation (gazebo) and Rviz. However i wanted to launch this robot only in Rviz without using gazebo. I tried following this tutorials of moviet humble: https://moveit.picknik.ai/main/doc/how_to_guides/moveit_configuration/moveit_configuration_tutorial.html and also this one: https://moveit.picknik.ai/main/doc/how_to_guides/moveit_launch_files/moveit_launch_files_tutorial.html

I tried to follow the way presented in the tutorials and this is my code: 

import os
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.substitutions import LaunchConfiguration, PathJoinSubstitution
from launch.conditions import IfCondition, UnlessCondition
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare
from launch.actions import ExecuteProcess
from ament_index_python.packages import get_package_share_directory
from moveit_configs_utils import MoveItConfigsBuilder
from lbr_description import LBRDescriptionMixin, RVizMixin
from lbr_bringup import LBRMoveGroupMixin

def generate_launch_description():

    declared_arguments = []
    declared_arguments.append(
        DeclareLaunchArgument(
            "rviz_config",
            default_value="miariiwa1.rviz",
            description="RViz configuration file",
        )
    )

    return LaunchDescription(
        declared_arguments + [OpaqueFunction(function=launch_setup)]
    )

def launch_setup(context, *args, **kwargs):

    moveit_config = (
        MoveItConfigsBuilder("iiwa14")
        .robot_description(os.path.join(
                get_package_share_directory("lbr_description"),
                "urdf/iiwa14/iiwa14.urdf.xacro",
            )
        )
        .trajectory_execution(file_path="/home/antonio/lbr-stack/src/lbr_fri_ros2_stack/lbr_moveit_config/iiwa14_moveit_config/config/moveit_controllers.yaml")
        .planning_scene_monitor(
            publish_robot_description=True, publish_robot_description_semantic=True
        )
        .planning_pipelines(
            pipelines=["ompl"]
        )
        .robot_description_kinematics(file_path="/home/antonio/lbr-stack/src/lbr_fri_ros2_stack/lbr_moveit_config/iiwa14_moveit_config/config/kinematics.yaml")
        .to_moveit_configs()   
    )

    # Start the actual move_group node/action server
    run_move_group_node = Node(
        package="moveit_ros_move_group",
        executable="move_group",
        output="screen",
        parameters=[moveit_config.to_dict()],
    )

    rviz_base = LaunchConfiguration("rviz_config")
    rviz_config = PathJoinSubstitution(
        #[FindPackageShare("lbr_description"), "config", rviz_base]
        [FindPackageShare("miar_robot"), "launch", rviz_base]
        
    )

    # RViz
    rviz_node = Node(
        package="rviz2",
        executable="rviz2",
        name="rviz2",
        output="log",
        arguments=["-d", rviz_config],
        parameters=[
            moveit_config.robot_description,
            moveit_config.robot_description_semantic,
            moveit_config.robot_description_kinematics,
            moveit_config.planning_pipelines,
            moveit_config.joint_limits,
        ],
    )

    # Static TF
    static_tf = Node(
        package="tf2_ros",
        executable="static_transform_publisher",
        name="static_transform_publisher",
        output="log",
        arguments=["0.0", "0.0", "0.0", "0.0", "0.0", "0.0", "world", "link_0"],
    )

    # Publish TF
    robot_state_publisher = Node(
        package="robot_state_publisher",
        executable="robot_state_publisher",
        name="robot_state_publisher",
        output="both",
        parameters=[moveit_config.robot_description],
    )

    # ros2_control using FakeSystem as hardware
    ros2_controllers_path = os.path.join(
        get_package_share_directory("lbr_ros2_control"),
        "config",
        "lbr_controllers.yaml",
    )
    ros2_control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[moveit_config.robot_description, ros2_controllers_path],
        output="both",
    )

    joint_state_broadcaster_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=[
            "joint_state_broadcaster",
            "--controller-manager-timeout",
            "300",
            "--controller-manager",
            "/controller_manager",
        ],
    )

    arm_controller_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=["joint_trajectory_controller", "-c", "/controller_manager"],
    )

    nodes_to_start = [
        rviz_node,
        static_tf,
        robot_state_publisher,
        run_move_group_node,
        ros2_control_node,
        joint_state_broadcaster_spawner,
        arm_controller_spawner,
    ]

    return nodes_to_star

PS: I put this launch file in a package called miar_robot as you can check in the code. However when i run this code in rviz i can plan a trajectory but i can´t execute it as you can see in the picture below:
[image]

In the terminal appears the following error: [ERROR] [ros2_control_node-5]: process has died [pid 15443, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_e0scg7kp --params-file /home/antonio/lbr-stack/install/lbr_ros2_control/share/lbr_ros2_control/config/lbr_controllers.yaml'].
[spawner-7] [INFO] [1719412041.691157785] [spawner_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1719412041.739105946] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist                    Anyone know any way to launch the robot lbr only rviz and also be able to plan and execute in the rviz without using any simulator as gazebo?

mhubii, 2024-06-26:
Hi @antonio1matos , looking good , happy to see the gripper is making good progress. Sorry, little busy until the weekend.

So the `controller_manager` is loaded as plugin into Gazebo. No Gazebo -> no `controller_manager`.

That said, there is a way to use fake hardware, no simulation. This feature is missing here right now. Thank you for opening the issue. Let's look into that

mhubii, 2024-07-06:
for reference, this might be relevant: https://control.ros.org/master/doc/ros2_control/hardware_interface/doc/mock_components_userdoc.html

antonio1matos, 2024-07-10:
Ok. Thank you @mhubii.
I will follow that link. Then i will give you feedback.

antonio1matos, 2024-07-23:
This was another question I wanted to confirm. In the laucnh file you have: sim.launch you are using the hardware: <plugin>gazebo_ros2_control/GazeboSystem</plugin> from the lbr_system_interface.xacro file, right? And if I didn't want to use a physical simulator like the gazebo, and wanted to test with a real robot, I would use the hardware plugin: <plugin>lbr_ros2_control::SystemInterface</plugin>, right? 

These parts of code in the lbr_system_interface.xacro file present in the images below (sensors, torques..) are all related to the hardware to test on the real robot, right? 
[image]
[image]
[image]
But for using the motion in gazebo i only needed this parts of code, right? (see images below) 
[image]
[image]

antonio1matos, 2024-07-23:
@mhubii  If you could clarify these doubts for me I would be grateful. Thanks

mhubii, 2024-09-10:
closing this particular issue as now supported via:

```shell
ros2 launch lbr_bringup mock.launch.py
```

Make sure to run RViz in a second terminal (if desired)

```shell
ros2 launch lbr_bringup rviz.launch.py
```

Trying to fix the end-effector issue in another topic.

Introduced in #190

closed 2024-09-10 by mhubii