lbr-stack/lbr_fri_ros2_stack #194: FRI Connection Quality Drops and Disconnects

OmidRezayof, 2024-07-01:
Hi @mhubii! It's me again :)
Hope you're doing okay. 
I have recently discovered the FRI connection quality drops during execution of LBRServer and it leads to disconnection of FRI. This happens even when the only code running is the lbr_bringup to connect robot the robot and sunrise to FRI.
I have attached the error when this happens. 
I would greatly appreciate any kind of solutions to this matter.
[image]

OmidRezayof, 2024-07-01:
So I'm using ROS Humble on Ubuntu 22.04 LTS Jammy. And no I'm not using a docker. 
Yes the codes are almost up to date with the latests commits on Humble branch.
Regarding the joint limits, the issue is, even when the robot is in a safe position and not in motion and no other codes are running, the FRI disconnects occasionally. So I don't think it is a joint limit issue?

mhubii, 2024-07-01:
okay makes sense. What control rate are you running the robot at?

OmidRezayof, 2024-07-02:
So, after trying a ton of different things, it looks like one of the LAN cables was dead inside and it was causing the connection to drop. Replacing that lan cable solved the problems (or at least looks like!). I will close this issue now. Thanks @mhubii again for your support everytime!