lbr-stack/lbr_fri_ros2_stack #198: Digital I/O signal activation

pala5, 2024-07-26:
Helo @mhubii.

I have a question that came up while using your packages ([lbr_frI_ros2_stack](https://github.com/lbr-stack/lbr_fri_ros2_stack) and [fri](https://github.com/lbr-stack/fri)).
I have the LBR Iiwa 14 820R with the Sunrise.OS 1.16.1.9 (FRI 1.16), and I am working with Ubuntu 22.04 and ROS2 Humble distro.   

We have installed an [external digital I/O card](https://my.kuka.com/s/product/digital-io-1616-npn/01t580000037Vy5AAE?language=en_US), configurable from Sunrise.Workvisual, required to control some external devices.
I have seen, that in the [FRIClientSDK](https://github.com/lbr-stack/fri/blob/fri-1.16/FRI-Client-SDK_Cpp.zip), for the FRI 1.16 version, in the _frilLBRCommand_ class, there is the function _setDigitalIOValue_, but I did not found this functionality in this package.

I am missing something or it was not considered? 
Could you provide some help regarding this issue?

mhubii, 2024-07-26:
Hi @pala5 , thanks for raising this issue. This is indeed missing right now

pala5, 2024-07-26:
Hi @mhubii, are you planning to work on it soon?

mhubii, 2024-07-26:
well this is an important feature that should be supported. How urgent is it on your end?

pala5, 2024-07-26:
Well, I could say it is a matter of life and death.

pala5, 2024-08-06:
Hey @mhubii!
Do you have time to work on that? 
Thanks

mhubii, 2024-08-14:
really occupied these days :(, I will give it a quick look tomorrow, hope this get you unstuck. Genuinely sorry

pala5, 2024-08-26:
Hey @mhubii, do not worry! Thank you very much for your effort.

pala5, 2024-09-19:
Hey @mhubii!
I don't want to be a bore, but do you have any update? 
Thanks

mhubii, 2024-12-16:
here is a list of things that need to be done

- [ ] Add I/O to IDL messages: https://github.com/lbr-stack/lbr_fri_idl
- [ ] Add I/O to `lbr_fri_ros2`: https://github.com/lbr-stack/lbr_fri_ros2_stack/tree/rolling/lbr_fri_ros2/src/interfaces
- [ ] Add I/O to `lbr_ros2_control`: https://github.com/lbr-stack/lbr_fri_ros2_stack/tree/rolling/lbr_ros2_control
- [ ] Add I/O to `robot_description`: https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/rolling/lbr_description/ros2_control/lbr_system_interface.xacro

TalentLeshen-DLUT, 2024-12-22:
@mhubii Hello, I have modified the project according to your method this week, you can see these changes in the fork below. 
 https://github.com/TalentLeshen-DLUT/lbr_fri_ros2_stack/tree/dev-humble-io
After I finished the modification, the project could be built with normal colcon, but when I connected the hardware, I found that I could not connect with it. I've tried for a long time, but I can't solve the problem. I think I need your help.

mhubii, 2025-01-04:
hi @TalentLeshen-DLUT. Apologize the late response. Seems you've been putting quite some effort! I am back online now. Would you mind working this through together?

TalentLeshen-DLUT, 2025-01-06:
@mhubii First let me wish you a belated Merry Christmas and a happy New Year. I can't imagine how happy I would be if I could get your help. I think I'd love for you to configure io with me. During your absence, I tried to modify lbr_fri_idl,lbr_fri_ros2 and lbr_ros2_control according to the method you gave. I created msg for analog io and digital io, and tried both modifying and creating analog_command files directly in position_command, which you can see in the following pages. (Correspondingly, I modified part of ros2_control)
https://github.com/TalentLeshen-DLUT/lbr_fri_ros2_stack/tree/dev-humble-io (In this project, I make changes directly in position_command,wrench_command, and torque_command)
https://github.com/TalentLeshen-DLUT/lbr_fri_ros2_stack/tree/dev-humble-io-2 (In this project, I created analog_command and modified digital_command)
In addition, I modified async_client, but the result is not satisfactory, they both have the same problem: miss state interfaces and miss command interfaces. I don't know what caused this problem, I guess it is that the hardware of lbr_bringup needs to be modified.
[image]

TalentLeshen-DLUT, 2025-01-06:
@mhubii I'm sorry that the above may confuse you because I used machine translation and didn't check it. In fact, in Project 1, I tried to configure IO directly in the position,wrench, and torque commands. In Project 2, I tried to configure IO files (analog command and digital command) . But they ran into the same problem :miss state interface and miss command interface. I guess I need to change the hardware part of lbr bringup, but I'm not sure.

mhubii, 2025-01-08:
no worries! Happy new year to you too! Very excited to work on this together. I am a little occupied today but accepted your invite and will try to raise issue / pull request there. Hope we can build a robust solution and merge changes back in.

TalentLeshen-DLUT, 2025-03-03:
Hi, @mhubii 

I have implemented part of the` io_command_controller` on ROS2. When I run the following command:

`ros2 launch lbr_bringup hardware.launch.py model:=iiwa14`
I can see a topic: `/lbr/io_command_controller/xxxx`.

Now, I believe the next step is to modify the` LBR_Server.java`. Could anyone provide some advice or guidance on how to proceed, so I can complete this task as quickly as possible?

Any help would be greatly appreciated!

TalentLeshen-DLUT, 2025-03-03:
I have successfully modified` LBRServer.java` and established a connection to the KUKA IIWA14 robot. However, when attempting to publish` boolean and analog messages`, I am unable to find the appropriate topics. I found the following topics:

```
- /lbr/boolean_command_topic
- /lbr/digital_command_topic
- /lbr/analog_command_topic
```

However, these are not what I need, as I cannot find topics related to `Speed and Start`. I suspect there may be an issue in my project setup, but I have not yet identified the root cause. I would appreciate any guidance to resolve this.

Any help would be greatly appreciated!

Project Link：
[https://github.com/TalentLeshen-DLUT/lbr_fri_ros2_stack/tree/dev-humble-io-2](url)

LBRSever.java:
```
package lbr_fri_ros2;

import static com.kuka.roboticsAPI.motionModel.BasicMotions.positionHold;

import java.util.Arrays;
import java.util.concurrent.TimeUnit;
import java.util.concurrent.TimeoutException;

import javax.inject.Inject;  

import com.kuka.roboticsAPI.applicationModel.RoboticsAPIApplication;
import com.kuka.roboticsAPI.controllerModel.Controller;
import com.kuka.roboticsAPI.deviceModel.LBR;

import com.kuka.roboticsAPI.uiModel.ApplicationDialogType;
import com.kuka.roboticsAPI.motionModel.controlModeModel.*;
import com.kuka.connectivity.fastRobotInterface.*;
import com.kuka.generated.ioAccess.MediaFlangeIOGroup;
import com.kuka.generated.ioAccess.SpindleIOGroup;

public class LBRServer extends RoboticsAPIApplication {
    // Robot and controller objects
    private LBR lbr_;
    private Controller lbr_controller_;

    // If dependency injection is supported, MediaFlangeIOGroup will be automatically injected
    @Inject
    protected SpindleIOGroup spindle;

    // Define control mode enum
    private enum CONTROL_MODE {
        POSITION_CONTROL,
        JOINT_IMPEDANCE_CONTROL,
        CARTESIAN_IMPEDANCE_CONTROL;
    }

    // Convert enum to a string array for user selection
    public static String[] getNames(Class<? extends Enum<?>> e) {
        return Arrays.toString(e.getEnumConstants()).replaceAll("^.|.$", "").split(", ");
    }

    // FRI parameter settings
    private String client_name_;
    private String[] client_names_ = {"172.31.1.148", "192.170.10.1"};
    private int send_period_;    
    private String[] send_periods_ = {"1", "2", "5", "10"};  // Send period in milliseconds

    // FRI-related objects
    private FRIConfiguration fri_configuration_;
    private FRISession fri_session_;
    private FRIJointOverlay fri_overlay_;

    // Motion control and command mode settings
[... 123 lines cut ...]
    @Override
    public void run() {
        // Hold the robot's current pose using the specified control mode, overlaying the FRI control layer
        lbr_.move(positionHold(control_mode_, -1, TimeUnit.SECONDS).addMotionOverlay(fri_overlay_));
    }
    
    @Override
    public void dispose() {
        // Close the FRI connection and release resources
        getLogger().info("Disposing FRI session.");
        fri_session_.close();
        super.dispose();
    }
    
    /**
     * Main entry point.
     * @param args Command line arguments.
     */
    public static void main(final String[] args) {
        LBRServer app = new LBRServer();
        app.runApplication();
    }
}

```

TalentLeshen-DLUT, 2025-03-08:
@mhubii 
I think I have achieved the configuration and use of IO interface, the following is the specific method:
This guide outlines the steps required to set up and use the IO interface on the robot controller via ROS2. Follow these four steps:

1. Create the IO Interface in WorkVisual

Define your IO interface in WorkVisual as needed for your application.

2. Modify `LBRServer.java` – Add Package Import

In `LBRServer.java`, add the necessary import statement to include the generated IO access:

`import com.kuka.generated.ioAccess.SpindleIOGroup;`

3. Modify LBRServer.java – Register IO Ports

In `LBRServer.java`, declare the IO group and register the IO ports. For example:

`protected SpindleIOGroup spindle;`

```
// Register IO ports
if (spindle != null) {
    fri_configuration_.registerIO(spindle.getOutput("Speed"));
    fri_configuration_.registerIO(spindle.getOutput("Start"));
    getLogger().info("MediaFlange IO registered.");
} else {
    getLogger().info("Spindle IO not available.");
}
```
This code ensures that the IO ports `"Speed"` and `"Start" `are registered with the configuration.

4. Modify `async_client.cpp` – Set IO Data Values

In `async_client.cpp`, add the following lines to send the IO data values to the robot controller:

```
robotCommand().setAnalogIOValue("Spindle.Speed", 1000);
robotCommand().setBooleanIOValue("Spindle.Start", true);
```
These commands set the analog IO value for `"Spindle.Speed"` and the boolean IO value for `"Spindle.Start"`.

By following these steps, you can successfully use the robot’s IO interface through ROS2. Feel free to include this guide in your repository’s documentation so that others can easily set up and use the IO functionality.

mhubii, 2025-03-12:
wow this is great @TalentLeshen-DLUT

TalentLeshen-DLUT, 2025-03-13:
@mhubii  I have modified the code so that the spindle can now be controlled by sending messages (two double values) to the `/lbr/command/spindle` topic, enabling both spindle startup and speed adjustments. I believe this successfully implements the necessary I/O additions. I'm happy to submit a pull request if needed.

TalentLeshen-DLUT, 2025-03-14:
I have submitted a PR in the hope that this work will help more people.

pala5, 2025-05-21:
Hi @mhubii,

Does the PR of @TalentLeshen-DLUT been validated and accepted?

mhubii, 2025-05-21:
hi @pala5, no sorry, haven't had the time to go through the PR unfortunately. But it should work