lbr-stack/lbr_fri_ros2_stack #199: iiwa14 with robotiq hand_e

Miftahur92, 2024-07-31:
I am trying to create a use a robotiq hande gripper with iiwa14. I have used the iiwa14 urdf file provided in this package. I have create robotiq hande description packages and control packages. When I run the robotiq hande gipper packages alone without attaching it to iiwa14 URDF, it runs perfectly fine (image attached)
[image]
But when I added the gripper URDF in the iiwa14 URDF, I can only see the gripper base link. Rviz shows an error that the there is no joint between the fingers and the gripper base (image attached)
[image]
 
This is the modified iiwa14 URDF:
`<?xml version="1.0"?>

<robot name="iiwa14" xmlns:xacro="http://www.ros.org/wiki/xacro">

    
    <xacro:include filename="$(find lbr_description)/urdf/iiwa14/iiwa14_description.xacro" />

    <xacro:arg name="robot_name" default="lbr" />
    <xacro:arg name="sim" default="true" />
    <xacro:arg
        name="system_parameters_path"
        default="$(find lbr_ros2_control)/config/lbr_system_parameters.yaml" />

    
    <link name="world" />

    
    <joint name="world_joint" type="fixed">
        <parent link="world" />
        <child link="link_0" />
    </joint>

    
    <xacro:iiwa14
        robot_name="$(arg robot_name)"
        sim="$(arg sim)"
        system_parameters_path="$(arg system_parameters_path)" />
    
    
  
    <xacro:include filename="$(find robotiq_hande_description)/urdf/robotiq_hande_gripper.urdf.xacro" />

    <link name="gripper_root_link">
    </link>

    <joint name="gripper_root_joint" type="fixed">
        <parent link="link_ee" />
        <child link = "gripper_root_link" />
        <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0" />
    </joint>

    
    <xacro:robotiq_hande_gripper prefix="" parent="gripper_root_link" >
        <origin xyz="0.0 0 0.005" rpy="0 0 0"/>
    </xacro:robotiq_hande_gripper>

</robot>`

Can I anyone please help me to debug the issues?

mhubii, 2024-09-08:
hi @Miftahur92 , the error suggests a missing transformation. Maybe the robot state publisher doesn't receive the joint states correctly?

mhubii, 2024-09-10:
related to #169