lbr-stack/lbr_fri_ros2_stack #201: Issues running the LBRServer code in Sunrise after adding EMS in safety configuration

SagarJoshi73249, 2024-08-10:
Thank you for your work, it's really helpful!

I am using the same code from the documentation but trying to add two EMS so that I can run the Hand Guiding mode in Kuka. I have gotten a basic code for hand guiding working and it works as intended but then I went back to the original LBRServer code with the EMS  one of them being enable hand guiding device inactive and the other to measure cartesian velocity and stop if it is over 1000mm/s. I want to integrate hand guiding in the LBRServer code so that I could collect pose and force data from the hand guided motions. Is there any way around for this? Thank you for reading.

mhubii, 2024-09-08:
hi @SagarJoshi73249. Sorry, got no experience with KUKA's hand-guiding. Is it possible to do hand-guiding and run the FRI connection simultaneously?

SagarJoshi73249, 2024-09-17:
Thanks for your reply. I don't think hand guiding and running the fri connection simultaneously is possible. This is mainly because of the safety monitoring options in sunrise. To get hand guiding to work, we need to add these ESM to sunrise but on adding these there are issues with getting the fri connection to work I don't know why.

i also tried implementing admittance control but that seems to be very janky and moving the robot is very awkward when compared to the smooth and straightforward motions with hand guiding.