lbr-stack/lbr_fri_ros2_stack #205: Timeout before FRI connection

Enzo-let, 2024-09-19:
I have set the ip like below.  And it can ping the ip of iiwa 14 robot by x66. But when I run the demo, it didn't work. About fri , I think it's 1.15.  I forgot to say i'm using the docker in ubuntu22.04(not VM). I run all them in docker.
I put the problem images below. I hope i can get your response. Thank you very much!!
[image]
[image]
[image]
[image]
[image]

mhubii, 2024-09-20:
hi @Enzo-let , thanks for the detailed pictures. Looks all good.

Docker should work well, but I'll double check. Did you follow these steps:

https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/docker/doc/docker.html

Enzo-let, 2024-09-20:
I really appreciate that you answer me. I have found  the problem and resolved it. It’s not the problem of the fire well. I checked the fri version again. I thought it’s 1.15. But i still tried to change the version of fri. And it worked. I don’t know why. Maybe it’s not 1.15. And i want to say thank you again. You have created a really good project.

mhubii, 2024-09-20:
happy to hear that! KUKA's versioning is not as reliable somehow

closed 2024-09-20 by mhubii