lbr-stack/lbr_fri_ros2_stack #206: Controller Manager not available

jclinton830, 2024-09-20:
Hi, 

I am currently running hardware.launch from the humble branch to bring up a real robot. I made sure to setup the robot's client as described in the doc. However, there seems to be a problem between the lbr_system_interface and ros2_control because even though the controller_manger node appears in the ros2 node list, the controller_manager is not available for the joint_state_broadcaster and the joint_trajectroy_controller to spawn properly. 

See the below log. 

```
~/lbr_ws$ ros2 launch lbr_bringup hardware.launch.py  model:=med14
[INFO] [launch]: All log files can be found below /home/jetsonadmin/.ros/log/2024-09-20-12-16-23-899873-jetsonadmin-desktop-87821
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [87830]
[INFO] [ros2_control_node-2]: process started with pid [87833]
[INFO] [spawner-3]: process started with pid [87837]
[INFO] [spawner-4]: process started with pid [87839]
[ros2_control_node-2] [INFO] [1726798584.504025510] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-2] [INFO] [1726798584.508064372] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-2] [INFO] [1726798584.509668172] [controller_manager]: Successful set up FIFO RT scheduling policy with priority 50.
[robot_state_publisher-1] [INFO] [1726798584.558961863] [robot_state_publisher]: got segment link_0
[robot_state_publisher-1] [INFO] [1726798584.559213734] [robot_state_publisher]: got segment link_1
[robot_state_publisher-1] [INFO] [1726798584.559235653] [robot_state_publisher]: got segment link_2
[robot_state_publisher-1] [INFO] [1726798584.559246181] [robot_state_publisher]: got segment link_3
[robot_state_publisher-1] [INFO] [1726798584.559255109] [robot_state_publisher]: got segment link_4
[robot_state_publisher-1] [INFO] [1726798584.559262949] [robot_state_publisher]: got segment link_5
[robot_state_publisher-1] [INFO] [1726798584.559270181] [robot_state_publisher]: got segment link_6
[robot_state_publisher-1] [INFO] [1726798584.559277349] [robot_state_publisher]: got segment link_7
[robot_state_publisher-1] [INFO] [1726798584.559284293] [robot_state_publisher]: got segment link_ee
[robot_state_publisher-1] [INFO] [1726798584.559291301] [robot_state_publisher]: got segment mount
[robot_state_publisher-1] [INFO] [1726798584.559298245] [robot_state_publisher]: got segment world
[robot_state_publisher-1] [INFO] [1726798584.559305029] [robot_state_publisher]: got segment zedm_camera_center
[robot_state_publisher-1] [INFO] [1726798584.559312485] [robot_state_publisher]: got segment zedm_camera_link
[robot_state_publisher-1] [INFO] [1726798584.559319205] [robot_state_publisher]: got segment zedm_left_camera_frame
[robot_state_publisher-1] [INFO] [1726798584.559326245] [robot_state_publisher]: got segment zedm_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1726798584.559333509] [robot_state_publisher]: got segment zedm_right_camera_frame
[robot_state_publisher-1] [INFO] [1726798584.559340517] [robot_state_publisher]: got segment zedm_right_camera_optical_frame
[ros2_control_node-2] [INFO] [1726798584.642479074] [controller_manager]: Received robot description file.
[ros2_control_node-2] [INFO] [1726798584.643544605] [resource_manager]: Loading hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1726798584.650571708] [resource_manager]: Initialize hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1726798584.650825947] [lbr_fri_ros2::AsyncClient]: Configuring client
[ros2_control_node-2] [INFO] [1726798584.650861051] [lbr_fri_ros2::AsyncClient]: Client command mode: 'POSITION'
[ros2_control_node-2] [INFO] [1726798584.650868923] [lbr_fri_ros2::AsyncClient]: Command guard variant 'default'
[ros2_control_node-2] [INFO] [1726798584.650943707] [lbr_fri_ros2::CommandGuard]: *** Parameters:
[ros2_control_node-2] [INFO] [1726798584.650955995] [lbr_fri_ros2::CommandGuard]: *   Joint A1 limits: Position: [-170.0, 170.0] deg, velocity: 85.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1726798584.650970651] [lbr_fri_ros2::CommandGuard]: *   Joint A2 limits: Position: [-120.0, 120.0] deg, velocity: 85.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1726798584.650978587] [lbr_fri_ros2::CommandGuard]: *   Joint A3 limits: Position: [-170.0, 170.0] deg, velocity: 100.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1726798584.650985435] [lbr_fri_ros2::CommandGuard]: *   Joint A4 limits: Position: [-120.0, 120.0] deg, velocity: 75.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1726798584.650992315] [lbr_fri_ros2::CommandGuard]: *   Joint A5 limits: Position: [-170.0, 170.0] deg, velocity: 130.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1726798584.650999034] [lbr_fri_ros2::CommandGuard]: *   Joint A6 limits: Position: [-120.0, 120.0] deg, velocity: 135.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1726798584.651006330] [lbr_fri_ros2::CommandGuard]: *   Joint A7 limits: Position: [-175.0, 175.0] deg, velocity: 135.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1726798584.651017626] [lbr_fri_ros2::JointPIDArray]: *** Parameters:
[ros2_control_node-2] [INFO] [1726798584.651023002] [lbr_fri_ros2::JointPIDArray]: *   p: 0.1
[ros2_control_node-2] [INFO] [1726798584.651029658] [lbr_fri_ros2::JointPIDArray]: *   i: 0.0
[ros2_control_node-2] [INFO] [1726798584.651034778] [lbr_fri_ros2::JointPIDArray]: *   d: 0.0
[ros2_control_node-2] [INFO] [1726798584.651039706] [lbr_fri_ros2::JointPIDArray]: *   i_max: 0.0
[ros2_control_node-2] [INFO] [1726798584.651044538] [lbr_fri_ros2::JointPIDArray]: *   i_min: 0.0
[ros2_control_node-2] [INFO] [1726798584.651049210] [lbr_fri_ros2::JointPIDArray]: *   antiwindup: false
[ros2_control_node-2] [INFO] [1726798584.651082778] [lbr_fri_ros2::StateInterface]: *** Parameters:
[ros2_control_node-2] [INFO] [1726798584.651088794] [lbr_fri_ros2::StateInterface]: *   external_torque_cutoff_frequency: 10.0 Hz
[ros2_control_node-2] [INFO] [1726798584.651095162] [lbr_fri_ros2::StateInterface]: *   measured_torque_cutoff_frequency: 10.0 Hz
[ros2_control_node-2] [INFO] [1726798584.651102970] [lbr_fri_ros2::AsyncClient]: Open loop 'true'
[ros2_control_node-2] [INFO] [1726798584.651109370] [lbr_fri_ros2::AsyncClient]: Client configured
[ros2_control_node-2] [INFO] [1726798584.716336906] [resource_manager]: Successful initialization of hardware 'lbr_system_interface'
[ros2_control_node-2] [INFO] [1726798584.716718057] [resource_manager]: 'configure' hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1726798584.716748489] [resource_manager]: Successful 'configure' of hardware 'lbr_system_interface'
[ros2_control_node-2] [INFO] [1726798584.716777960] [resource_manager]: 'activate' hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1726798584.716819880] [lbr_fri_ros2::App]: Opening UDP socket with port_id '30200'
[ros2_control_node-2] [INFO] [1726798584.716903336] [lbr_fri_ros2::App]: Socket opened successfully
[ros2_control_node-2] [INFO] [1726798584.717133991] [lbr_ros2_control::SystemInterface]: Awaiting robot heartbeat. Attempt 1, remote_host 'INADDR_ANY', port_id '30200'
[ros2_control_node-2] [INFO] [1726798584.717278790] [lbr_fri_ros2::App]: Realtime kernel recommended but not required
[ros2_control_node-2] [INFO] [1726798584.717298182] [lbr_fri_ros2::App]: Starting run thread
[ros2_control_node-2] [INFO] [1726798584.727023673] [lbr_fri_ros2::AsyncClient]: LBR switched from 'IDLE' to 'MONITORING_WAIT'
[... 12 lines cut ...]
[ros2_control_node-2] [INFO] [1726798591.719026635] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'MONITORING_READY'.
[ros2_control_node-2] [INFO] [1726798592.719331187] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'MONITORING_READY'.
[ros2_control_node-2] [INFO] [1726798593.719607826] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'MONITORING_READY'.
[ros2_control_node-2] [INFO] [1726798594.719830536] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'MONITORING_READY'.
[spawner-4] [WARN] [1726798595.070440935] [spawner_joint_trajectory_controller]: Could not contact service /controller_manager/list_controllers
[... cut ...]
[spawner-4] [INFO] [1726798605.095798514] [spawner_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-3] [WARN] [1726798605.139529150] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[spawner-3] [INFO] [1726798605.140469689] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-2] [INFO] [1726798605.722598962] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'MONITORING_READY'.

```

mhubii, 2024-09-20:
hi @jclinton830. So the connection is established well. The driver awaits `COMMANDING_WAIT` state.

To achieve this, simply press the green play button whilst holding one of the gray enabling switches on the KUKA smartPAD.

In automatic mode, the robot will connect immediately.

jclinton830, 2024-09-20:
> hi @jclinton830. So the connection is established well. The driver awaits `COMMANDING_WAIT` state.
> 
> To achieve this, simply press the green play button whilst holding one of the gray enabling switches on the KUKA smartPAD.
> 
> In automatic mode, the robot will connect immediately.

Hi @mhubii I am running the robot in automatic mode. 
My question is not about the awaiting 'COMMANDING WAIT'

I want to know why the controller manager is not available for the joint state broadcaster and the joint trajectory controller to spawn?

mhubii, 2024-09-20:
the controller manager is stuck loading the `lbr_ros2_control::SystemInterface` since the robot never enters `COMMANDING_WAIT` state. This likely happens because the robot is in test mode `T1`

jclinton830, 2024-09-24:
Thanks for the explanation. I can confirm this is now working.

closed 2024-09-24 by jclinton830