lbr-stack/lbr_fri_ros2_stack #217: Impetence control

zzrobot1979, 2024-10-31:
Hello: I would like to know whether impedance control can be achieved in this project, or can it be achieved directly using the wrench method?

mhubii, 2024-10-31:
hi @zzrobot1979, yes, you can do impedance control. In impedance control, do you wish to control positions or torques?

zzrobot1979, 2024-11-01:
Thank you very much for your reply! I want to make the end of my kuka robot arm move smoothly along a certain space vector under a constant contact force. Maybe I should control the speed and force/torque of the end of the robot arm at the same time?

mhubii, 2024-11-01:
hope this demo gets you up to speed:

https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_py/doc/lbr_demos_py.html#lbr-wrench-command-controller-hardware-only

zzrobot1979, 2024-11-02:
Thank you very much for your reminder! I would like to know how to calibrate the quality of the end effector of the robot arm? Is this process configured in sunrise or can it be done by changing some parameters of this project?

zzrobot1979, 2024-11-02:
Another question is: how to realize the movement of the robot in lbr_demos/lbr_demos_cpp/src/wrench_sine_overlay.cpp? Is it controlled by the line of code lbr_wrench_command_.joint_position = lbr_state_.measured_joint_position;? How to pass a position in space to the robot? Thank you very much!

mhubii, 2024-11-02:
> Thank you very much for your reminder! I would like to know how to calibrate the quality of the end effector of the robot arm? Is this process configured in sunrise or can it be done by changing some parameters of this project?

yes, the load data calibration needs to be done on the smart pad, can try to provide some instructions

> Another question is: how to realize the movement of the robot in[ lbr_demos/lbr_demos_py/lbr_demos_py/wrench_sine_overlay.py](url)? Is it controlled by the line of code ` self._lbr_wrench_command.joint_position = deepcopy( self._lbr_state.measured_joint_position )`? How to pass a position in space to the robot? Thank you very much!

that's correct. The target joint position is set exactly there (`self._lbr_wrench_command.joint_position`). If you wish to command some end-effector velocity, you'd have to do the kinematics yourself currently.

zzrobot1979, 2024-12-17:
Thank you for your answer. I have another question. Is it necessary to purchase the FRI expansion package of KUKA to realize your team's project? Or is the expansion package the link under your repository?  [https://github.com/lbr-stack/fri]

mhubii, 2024-12-17:
yes, it is a paid option. That said, often it is purchased. In the documentation https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_fri_ros2_stack/doc/hardware_setup.html#install-applications-to-the-robot under `5.2. Select FRI and software examples`, if this shows up you should have it