lbr-stack/lbr_fri_ros2_stack #219: lbr_moveit_cpp documentation out of data; unable to run hello_moveit demo

bluewish07, 2024-11-10:
https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_moveit_cpp/doc/lbr_moveit_cpp.html

The command for launching move_group seems to be out of date on this page. Nonetheless, I tried to run the move_group.launch.py which is now moved into individual folders under lbr_moveit_config and I still couldn't get the hello_moveit demo to run successfully. Specifically, hello_moveit.cpp seems to stuck at the MoveGroupInterface initialization.

Could you advise? 
branch: humble

mhubii, 2024-11-16:
Hi @bluewish07. Thank you for raising this issue. You are right, this demo is probably outdated. Are you working on actual hardware or in simulation?

bluewish07, 2024-11-16:
Hi @mhubii thank you for getting back to me! Eventually I found out that I should run move_group with lbr_bringup and it did work. So I think just updating the doc on that page would be great. Thank you for your help!