lbr-stack/lbr_fri_ros2_stack #221: Engaging Robot's Brakes

OmidRezayof, 2024-11-19:
Hi @mhubii !
Hope you're doing well.
I have experienced the robot (med 7) giving error about current overloading due to long period of use.
I contacted KUKA and they said I should engage the physical breaks of the robot whenever the robot is not moving for a while and disengage them before moving. I have been doing this manually by pressing the e-stop button. I know that Sunrise have built-in functions in Java which we can do this through codes, but I was wondering if lbr-stack has a solution for this.

mhubii, 2024-11-19:
hi @OmidRezayof , that's interesting. How long did you run the robot for? The only way to engage the brakes via the FRI (at least with version 1.x) is to shut the connection.

OmidRezayof, 2024-11-21:
I'm having a payload of 3Kgs on the robot, and if I don't engage the brakes for more than 2-3 hours, it complains about current overloading. 
Not that I necessarily want to use the robot for that long, but after talking to the KUKA support people, I realized it is a healthy habit to engage the brakes more often when the robot is not supposed to move. And I wanted to see if lbr-stack has a way to do so : )

caioaamaral, 2025-07-08:
what about pausing robot application while  Command mode is active?

> If the robot application is paused while Command mode is active (for example by pressing the STOP key), the FRI client will not accept any further command specifications. The motion is stopped with path-oriented
braking and FRI switches to the state MONITORING_READY.

caioaamaral, 2025-07-14:
Pardon me, I'm not all familiar with FRI. But reading from the documentation it seems to be a matter of simply 'pausing/unpausing' the robot application, not necessarly killing the connection (which would put it at IDLE while pausing put it in MONITORING_READY).

[image]