lbr-stack/lbr_fri_ros2_stack #226: Joint Positions Not Updating in Real Time

anselmo-parnada, 2024-11-25:
Hi everyone,

For my application, I need to use the real-time joint positions of the robot to feed it into my custom controller.

Upon trying all the sine overlay demos whilst echoing the LBRState in the ROS2 CLI, I found that that measured joint positions do not update in real time. The joint positions printed remain the same even though the robot is performing an oscillating motion. The joint positions will only update when I release the enable key on the SmartPad to halt the LBRServer program that was running on the controller. 

Ideally, I would like to have these update with the motion. Oddly enough, the joint efforts are updated in real time, so it does not seem to be a connection issue.

Can somebody please help me with issue? Thank you in advance!

Some hardware/sofware specifications, for reference:

- Robot: KUKA IIWA 14
- FRI SDK Version: 1.11

mhubii, 2024-11-26:
hi @anselmo-parnada. Thank you for reporting this issue. This is quite odd indeed

anselmo-parnada, 2024-11-26:
Hi @mhubii,

I looked into it yesterday and have fixed the issue. In [line 63 in state.cpp](https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b19e7e22f9b5c2251b1028d366e2450689c1bba9/lbr_fri_ros2/src/interfaces/state.cpp#L63C1-L64C63), the measured joint position is copied to memory like so within `StateInterface::set_state_open_loop()`:

```cpp
std::memcpy(state_.measured_joint_position.data(), joint_position.data(),
              (double) * fri_state_t::NUMBER_OF_JOINTS);
```

But in [line 23 of state.cpp ](https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b19e7e22f9b5c2251b1028d366e2450689c1bba9/lbr_fri_ros2/src/interfaces/state.cpp#L23C3-L23C48), the measured joint position is copied to memory in a different way within `StateInterface::set_state()`:

```cpp
std::memcpy(state_.measured_joint_position.data(), state.getMeasuredJointPosition(),
              sizeof(double) * fri_state_t::NUMBER_OF_JOINTS);
```

I'm no expert on your codebase, but it seems `StateInterface::set_state()` is the function that is called at the beginning and end of the FRI connection and `StateInterface::set_state_open_loop()` is the function called every cycle during the FRI connection. I came to this conclusion as the line in `StateInterface::set_state()` is the way you would obtain joint position data from the robot using the FRI Client SDK from KUKA, which I confirmed worked on my machine with KUKA's demo FRI clients. The line in `StateInterface::set_state_open_loop()` was not something I'd seen used in the FRI client SDK, so I was suspicious it was the cause of the issue.

Based on this assumption, I changed [line 63 in state.cpp](https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b19e7e22f9b5c2251b1028d366e2450689c1bba9/lbr_fri_ros2/src/interfaces/state.cpp#L63C1-L64C63) to match [line 23 of state.cpp ](https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b19e7e22f9b5c2251b1028d366e2450689c1bba9/lbr_fri_ros2/src/interfaces/state.cpp#L23C3-L23C48) and it seemed to allow the joint positions to be updated in real time!

Not sure if this because of my version of FRI or ROS2, but it seemed to work. I think the `joint_position` argument going into `StateInterface::set_state_open_loop()` is not storing up-to-date data, which is why this issue occurs for me.

Moght be worth looking into. Hope this helps!

anselmo-parnada, 2024-11-26:
From my experience, I do not think the issue is tied to the control mode. This issue occured regardless of the control mode chosen, as I experienced the issue on all the sine overlay demos within the lbr_demos_py package. I ensured the the client's yaml files and LBRServer application we configured according to the documentation, too.

anselmo-parnada, 2024-11-26:
Fore reference, I did first notice this issue when I was playing around with the wrench_sine_overlay.py demo. I'm not sure if that helps narrow things down.

mhubii, 2024-12-14:
hi @anselmo-parnada. This issue is now fixed, so should the drifting you observed.

I do want to apologize for the bug and hope it did not cause any damage. I do my best to support all different cases but am doing this mostly in my spare time. Do keep in mind to always execute in `T1` mode first. Thank you for reporting this issue. I'll try to document the load data calibration #235 ASAP.

closed 2024-12-14 by mhubii