lbr-stack/lbr_fri_ros2_stack #230: Gazebo world link

mhubii, 2024-11-27:
The recent updates to the links (for dual arm support #189):

- #213
- #225

 cause a missing connection to Gazebo world, see below:

[image]

Hydran00, 2024-12-27:
Hi @mhubii, I had the same problem. I solved it by adding a joint from world to to the floating link. 
I just added the following code [here](https://github.com/idra-lab/lbr_fri_ros2_stack/blob/fdeb288f7897c23fea4fc9ccd367a7f693d60cfa/lbr_description/urdf/iiwa14/iiwa14.xacro#L17-L21
) in the `.xacro` definition.
```
  <link name="world" />
  <joint name="$world_to_base" type="fixed">
      <parent link="world" />
      <child link="$(arg robot_name)_floating_link" />
  </joint>
```
here

I think gazebo need a `world` link in the urdf to understand how the robot is connected to the world. Maybe the `world_robot_tf` in `gazebo.launch.py` were ignored since `world` did not exist?

mhubii, 2025-01-04:
hi @Hydran00 and thank you for the hint there. Happy new year!

closed 2025-03-26 by mhubii