lbr-stack/lbr_fri_ros2_stack #232: Torque control not working

Hydran00, 2024-12-02:
Hi,

thank you for your work!

We managed to install everything smoothly and the position control demo worked fine, but encountered an issue during the torque control demos: the robot did not move. 

After some trials, we managed to make the torque demo work by adding the following line [here](https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/01dadf077f298c38deda45d02f19471e1d7a7536/lbr_fri_ros2/src/interfaces/torque_command.cpp#L38)
```
command_.torque = command_target_.torque;
```

It is very strange, since there are no open or closed issues reporting this problem. Do you have any suggestion?

mhubii, 2024-12-03:
hi @Hydran00 , no worries. Quick question, did you set the `client_command_mode` to `torque` here:

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/01dadf077f298c38deda45d02f19471e1d7a7536/lbr_description/ros2_control/lbr_system_config.yaml#L6

There is currently an issue with forwarding the joint states in `torque` / `wrench` command mode, #226

Hydran00, 2024-12-03:
Thank you for you reply.
I followed the guide, therefore I correctly set the send period and command mode to torque.

mhubii, 2024-12-03:
okay I see, this is a bug then introduced in #225

Hydran00, 2024-12-03:
Ok, I will do it soon 👍

Hydran00, 2024-12-04:
By the way, we were trying to implement our own cartesian impedance controller to be able to change the stiffness achieving variable impedance control by using the robot in gravity compensation. To achieve gravity compensation, we set every stiffness gains to zero in the `LBRServer.java` [here](https://github.com/lbr-stack/fri/blob/e57df906b14f6af1306e5ea9e13a97d7bff51c09/server_app/LBRServer.java#L82).

In gravity compensation we were facing an issue on the pendant called `Illegal axis delta detected` and we managed to bypass it by setting the parameter `open_loop` to false in your interface.

The problem we were facing when sending zero torque to this hardware interface is that the robot has a strange behaviour: it seems to drift a bit even when no external forces are applied. 
Also, it has a lot of joint friction and poor tracking performances when we add our overlay torque for the impedance tracking of a trajectory. 

Do you think know other way to achieve gravity compensation?

You can check [our repository](https://github.com/lucabeber/effort_controller/tree/kuka) if you are interested in the implementation of the controller.

mhubii, 2024-12-04:
happy to try the controller. I haven't encountered `Illegal axis delta detected` before. Could you share a quick readme so I can try run it?

Hydran00, 2024-12-04:
Ok, I will update the readme so that you can replicate as soon as I can. 👌 

What do you mean by "the inverse dynamics model isn't perfect"? You mean that the FRI does not provide an accurate dynamics model?

Also, you mentioned that we should calibrate the load data, how can we achieve that? Do you mean by using the robot mastering button on the teach pendant? We do not have tools attached, but maybe we can try calibrating the sensors anyway.

Hydran00, 2024-12-04:
We created a [repository](https://github.com/idra-lab/kuka_impedance) to implement the Cartesian impedance control, with a small guide on how to make it work. As I told you, I cannot test it on the hardware, but it was working last week, even if with poor perfomances.

mhubii, 2024-12-04:
great, will check this out. Thank you again for raising the issue and fixing it! Going to add some doc for calibrating load data soon. Closing this issue for now

closed 2024-12-04 by mhubii

mhubii, 2024-12-11:
no worries @anselmo-parnada, I think it is important to have this discussion and I can confirm that the LBR, when controlled with zero stiffness, drifts and potentially overshoots. Two issues (imho):

1. Load data calibration. If load data isn't calibrated, the LBR's inverse dynamics model is off (it doesn't know anything is attached to it). The documentation for how to do this is missing #235. This should and will be added.
2. The inverse dynamics model of the LBR is not perfect. I am not sure how to address this, or whether this is something KUKA needs to improve.

Hydran00, 2024-12-11:
Hi @anselmo-parnada,

me and my team tried the gravity compensation with the robot in its mechanical zero position and our robot does not drift, however as soon as we move it, it starts drifting.
If you are facing the `Illegal axis delta` error, you can turn off `open_loop`, i.e. (`open_loop: false`) in the `lbr_system_config.yaml`, but the robot behaviour seems to change: it looks like there is a stiction force on each joint that prevents it from being smooth. 

You can easily check this by trying moving the robot base (`joint A1`) in the two different case. Please tell me if you have the same behaviour. 🙂

anselmo-parnada, 2024-12-11:
Hey @mhubii, thanks for the response. 

> 1. Load data calibration. If load data isn't calibrated, the LBR's inverse dynamics model is off (it doesn't know anything is attached to it). The documentation for how to do this is missing [Document load data calibration #235](https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/235). This should and will be added.

By calibrating the load data, are you referring to the tool load calibration program accessible via the SmartPad? I have so far only been testing the robot without a tool attached to it's mounting flange, so there is no load to calibrate.  Are you suggesting calibrating the load of an empty tool to "correct" the dynamics model?

I tried this before and gotten very strange values from calibration program. At best, I think this would "correct" the dynamics model only for joint configuration that you start the calibration process. However, this will not solve the underlying issue.

> 2\. The inverse dynamics model of the LBR is not perfect. I am not sure how to address this, or whether this is something KUKA needs to improve.

Perhaps they do not calibrate the dynamics for each unique robot they produce, so the dynamics model will always be inaccurate. Perhaps they do, and the dynamic properties of the robot gradually deviates from the model over time due to wear-and-tear. How old are the robots you guys are using?

anselmo-parnada, 2024-12-11:
Hi @Hydran00,

> me and my team tried the gravity compensation with the robot in its mechanical zero position and our robot does not drift, however as soon as we move it, it starts drifting.

Just tried this 3 times. The first time, it remained stationary and only began to drift when I moved it, similar to you. The next 2 times, it began drifting immediately after the position hold command was initiated. 

Does your robot like to drift in a particular direction? Our robot's `A1` joint likes to turn clockwise (looking from above), exclusively. Wonder if your robot does the same.

> If you are facing the `Illegal axis delta` error, you can turn off `open_loop`, i.e. (`open_loop: false`) in the `lbr_system_config.yaml`, but the robot behaviour seems to change: it looks like there is a stiction force on each joint that prevents it from being smooth.

I am already using `open_loop: false`, so I've not been having that issue. When do you experience the stiction force issue? I'm not clear what steps to follow to reproduce this issue. Please let me know, thanks!

Hydran00, 2024-12-11:
Our robot tends to have strange unwanted motion for joints `A4` and `A5` but not in `A1`. They also rotate every time toward the same direction as in your case.

However, we solved the stiction problem: it was due to the fact that we commented these lines in our fork. 
https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/d6aee144ac3fd71464686999272b29f4647bb23c/lbr_fri_ros2/src/interfaces/torque_command.cpp#L34-L37

We tried again our [cartesian impedance control](https://github.com/idra-lab/kuka_impedance) and this time the behaviour was much better. Even though the gravity compensation (i.e. stiffness set to zero) is not good, this errors are compensated by the torque contributes generated by the position error. If you want, you can give it a try. 🦾

mhubii, 2024-12-13:
> By calibrating the load data, are you referring to the tool load calibration program accessible via the SmartPad?

yes, been referring to that.

> How old are the robots you guys are using?

The robot that we are using are about 5 years old.

> However, we solved the stiction problem: it was due to the fact that we commented these lines in our fork.

When this line is removed, the target position remains unchanged, maybe that stops the observed drifting.

> If you want, you can give it a try. 🦾

Will try out this controller over the weekend. Hopefully I can replicate the issues.

RLi43, 2025-01-10:
> Are you suggesting calibrating the load of an empty tool to "correct" the dynamics model?

Hello @anselmo-parnada , I guess it will help. I tested with LBR Med 7, and the empty media flange was measured as around 0.5kg.