lbr-stack/lbr_fri_ros2_stack #242: The error occurs when running the commands in Quick Start 4.

abaicoming, 2024-12-17:
I tried running the commands in Quick Start 4, but encountered the following error：
`kukarobot@kukarobot-System-Product-Name:~/lbr-stack$ ros2 launch lbr_bringup mock.launch.py model:=med7
[INFO] [launch]: All log files can be found below /home/kukarobot/.ros/log/2024-12-17-19-06-09-586033-kukarobot-System-Product-Name-2927045
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable '[<launch.substitutions.text_substitution.TextSubstitution object at 0x726f5d9a8110>]' not found on the PATH`
Does running this command require connecting a real robot and successfully configuring its hardware?

mhubii, 2025-01-05:
hi @abaicoming, apologize the late response. Sorry, I cannot replicate this issue. There is an issue with invalid joint names #244, but it seams unrelated to your issue. Might there be an issue with the simulation in your case?

abaicoming, 2025-01-05:
Thank you very much for your reply. Yes, #244 and I are using the same demo, but we encountered different issues. Below, I will provide more detailed terminal output:

The following issues appeared in Terminal 1:
```
kukarobot@kukarobot-System-Product-Name:~$ ros2 launch lbr_bringup gazebo.launch.py     ctrl:=joint_trajectory_controller     model:=med7 # [iiwa7, iiwa14, med7, med14]
[INFO] [launch]: All log files can be found below /home/kukarobot/.ros/log/2025-01-05-23-34-15-908216-kukarobot-System-Product-Name-1352845
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [1352853]
[INFO] [robot_state_publisher-2]: process started with pid [1352854]
[INFO] [gazebo-3]: process started with pid [1352855]
[INFO] [parameter_bridge-4]: process started with pid [1352856]
[INFO] [create-5]: process started with pid [1352857]
[INFO] [spawner-6]: process started with pid [1352859]
[INFO] [spawner-7]: process started with pid [1352860]
[robot_state_publisher-2] [INFO] [1736091256.156636198] [lbr.robot_state_publisher]: Robot initialized
[static_transform_publisher-1] [INFO] [1736091256.157346264] [static_transform_publisher_Ye7Zlw1PbnJiIlih]: Spinning until stopped - publishing transform
[static_transform_publisher-1] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-1] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-1] from 'world' to 'lbr_floating_link'
[create-5] [INFO] [1736091256.177098570] [lbr.ros_gz_sim]: Requesting list of world names.
[parameter_bridge-4] [INFO] [1736091256.180125897] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (gz.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[spawner-6] [INFO] [1736091256.346239090] [lbr.spawner_joint_state_broadcaster]: waiting for service /lbr/controller_manager/list_controllers to become available...
[spawner-7] [INFO] [1736091256.346243358] [lbr.spawner_joint_trajectory_controller]: waiting for service /lbr/controller_manager/list_controllers to become available...
[create-5] [INFO] [1736091256.642109913] [lbr.ros_gz_sim]: Waiting messages on topic [robot_description].
[create-5] [INFO] [1736091256.740717129] [lbr.ros_gz_sim]: Entity creation successfull.
[INFO] [create-5]: process has finished cleanly [pid 1352857]
[gazebo-3] libEGL warning: egl: failed to create dri2 screen
[gazebo-3] libEGL warning: egl: failed to create dri2 screen
[gazebo-3] libEGL warning: egl: failed to create dri2 screen
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A1"]/physics/ode/provide_feedback:<urdf-string>:L0]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A2"]/physics/ode/provide_feedback:<urdf-string>:L0]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A3"]/physics/ode/provide_feedback:<urdf-string>:L0]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A4"]/physics/ode/provide_feedback:<urdf-string>:L0]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A5"]/physics/ode/provide_feedback:<urdf-string>:L0]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A6"]/physics/ode/provide_feedback:<urdf-string>:L0]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A7"]/physics/ode/provide_feedback:<urdf-string>:L0]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] (2025-01-05 23:34:18.242) [error] [SystemLoader.cc:91] Failed to load system plugin [gz_ros2_control-system] : Could not find shared library.
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A1"]/physics/ode/provide_feedback:<data-string>:L73]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A2"]/physics/ode/provide_feedback:<data-string>:L149]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A3"]/physics/ode/provide_feedback:<data-string>:L225]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-3] Warning [Utils.cc:132] [/sdf/model[@name="med7"]/joint[@name="lbr_A4"]/physics/ode/provide_feedback:<data-string>:L301]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[... cut ...]
In fact, I do have a KUKA Med7, but I am unable to launch the LBRServer application on the Smart Pad, as shown in the picture below. 
[image]
This is because I am encountering the same issue as #17. 
[image]
The person who raised this issue said:  "I am not familiar with Java, but we followed the automatic instructions of the IDE and it works."  And then they closed the issue. However, I am still facing this problem.
By the way my Ubuntu system is 24.04, and my ROS2 version is Rolling. I apologize for the trouble caused by my being a beginner with ROS2. I look forward to your reply.

mhubii, 2025-01-05:
no worries, hope I'll be able to help. I re-opened #17. Regarding Gazebo, maybe something changed on the `rolling` branch (the `rolling` branch the unstable branch for ROS). Chances are something is out of date. I'll have a look.

abaicoming, 2025-01-09:
Thank you very much for your response. I understand that the latest system versions always come with various compatibility issues, but as the saying goes, curiosity killed the cat. When I downloaded it, I just wanted to give the latest system a try. Fortunately, I have a physical robotic arm, and thanks to KUKA's excellent safety design, conducting experiments in T1 mode is safe enough without needing to run them in a simulation environment first. Moreover, I always have the last resort of reinstalling Ubuntu and the ROS 2 system and using a more stable version.

This is why resolving issue #17 is more critical for me, as I really need to get this expensive machine running to conduct some experiments. Unfortunately, after several days of effort, I still haven't been able to resolve #17. I’ve tried asking questions under issue #17 and even emailed the original poster, but I haven’t received any response. I also tried uninstalling and reinstalling the `SunriseWorkbenchMed `software on Windows and configuring it on another device, but none of these worked.

I suspect that the problem might be related to the version of `SunriseWorkbenchMed`, as I am using version `2.7.0`, while the documentation mentions version `1.15`. I have already contacted the engineer who sold us the KUKA arm to request version `1.15`. Once I get it, I will give it a try and provide feedback to you.

The error in `LBRServer.java` indicates that some library files cannot be found. I'm not sure if this is due to an environment issue caused by using the wrong version of `SunriseWorkbenchMed `or if I need to import additional `JAR `files.

```
Multiple markers at this line
	- The import com.kuka.roboticsAPI.controllerModel.Controller cannot be 
	 resolved
	- The import com.kuka.roboticsAPI.deviceModel.LBR cannot be resolved
	- The import com.kuka.connectivity cannot be resolved
```

[image]

If these errors give you any ideas on how to resolve the issue, please let me know. I would greatly appreciate it.

mhubii, 2025-01-09:
apologize the inconvenience. Since you are using version `2.7`, may I ask whether you put the correct `LBRServer.java` onto the KUKA controller? You can find it here: https://github.com/lbr-stack/fri/blob/fri-2.7/server_app/LBRServer.java

abaicoming, 2025-01-14:
## Thank you very much for your reminder. 
Since the version of `lbr_fri_ros2_stack` used in Ubuntu is `1.15`, I subconsciously used version `1.15` of `LBRServer.java`. I have now updated both `lbr_fri_ros2_stack `and `LBRServer.java` to version `2.7`, and downgraded the versions of Ubuntu and ROS2 to `22.04` and `Humble`. After a few days of testing, all the demos are running smoothly. I greatly appreciate your helpful answers.
  
By the way, when running `lbr_demos_advanced_cpp -> Admittance Controller`, I encounter the following error:
```
[ros2_control_node-3] [ERROR] [1736781170.176570586] [lbr.admittance_controller]: The admittance controller currently requires custom lbr_system_config.yaml configurations. 
Therefore, only experienced users should use it, then remove this error and strictly follow https://lbrstack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_advanced_cpp/doc/ lbr_demos_advanced_cpp.html#admittance-controller. 
This error can be removed when a) the controller works with default system configurations and b) the controller checks that load data was successfully calibrated.
[ros2_control_node-3] [ERROR] [1736781170.176627369] [lbr.controller_manager]: Could not initialize the controller named 'admittance_controller'
[ros2_control_node-3] [INFO] [1736781170.187891566] [lbr.controller_manager]: Loading controller 'lbr_state_broadcaster'
[spawner-7] [FATAL] [1736781170.189059632] [lbr.spawner_admittance_controller]: Failed loading controller admittance_controller
```
Is it necessary to perform the operation on the smart pad, as mentioned in #217 [https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/217#issuecomment-2453021393] , to load the data calibration? Fortunately, running `lbr_demos_advanced_py -> Admittance Controller` was successful, and it also can provide me with admittance control. By the way, before launching it, I need to install the optas library; otherwise, I get the following error:
```
File "/home/kukamed7/lbr-stack/build/lbr_demos_advanced_py/lbr_demos_advanced_py/ admittance_controller.py", line 2, in <module>
import optas
ModuleNotFoundError: No module named 'optas
```
It seems that this is mentioned in #42.

## Finally, there are a few things that are bothering me, and I would like to ask:
### 1.	In _lbr_demos_advanced_cpp -> Twist Controller_
The target being tracked is a 6D force in Cartesian space:
```
ros2 topic pub \
    --rate 100 \
    /lbr/command/twist \
    geometry_msgs/msg/Twist \
    "{linear: {x: 0.0, y: 0.0, z: 0.05}, angular: {x: 0.0, y: 0.0, z: 0.0}}"
```
If the external force reaches the target force, the manipulator should theoretically stop. However, during testing, I found that it does not.
 
 ### 2.	_lbr_demos_py -> LBR Wrench Command Controller (Hardware only)_
After starting the code, I observed that the robot's end effector does indeed move with a small sinusoidal displacement in the XY plane. Based on the name, I initially thought it was just a force control program, but after checking the code, I found it’s not the case. I noticed that the `LBRWrenchCommand.msg` data type includes both joint_position and wrench, and in the code, joint_position is set to the robot's initial joint position, while wrench is set to a sinusoidal signal. Does this controller enable simultaneous tracking of joint_position and 6D end-effector force, similar to an impedance control with a spring-like effect?
 
 ### 3.	_lbr_demos_advanced_py -> admittance_controller.py_
I looked at the code and found that the implementation approach is roughly as follows:
$f_{ext} = J^{-1} \tau_{ext}, \quad \Delta x = f(f_{ext}), \quad \Delta q = \alpha J^{-1} \Delta x$
What I'm curious about is that it seems possible to achieve the same effect through the `"Torque Command Controller"` by controlling the joint torque of the manipulator to make **$\tau_{ext}$**. In other words, make the joint torque track the magnitude of the torque before the external force is applied. This approach also appears to be simpler.

**Apologies for the lengthy message. If you have some time, could you kindly take a look? Once again, I truly appreciate your enthusiastic help.**

mhubii, 2025-01-16:
happy to hear things are running (mostly) now @abaicoming.

> By the way, when running `lbr_demos_advanced_cpp -> Admittance Controller`, I encounter the following error:

You can comment this line

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_ros2_control/src/controllers/admittance_controller.cpp#L50

to run the admittance control. This line was added because load data calibration is currently not documented. If a weight is attached to the end-effector, this may currently cause undesired behavior. Apologies for that.

> If the external force reaches the target force, the manipulator should theoretically stop. However, during testing, I found that it does not.

To my best understanding, a twist describes a velocity, not a force.

> Does this controller enable simultaneous tracking of joint_position and 6D end-effector force, similar to an impedance control with a spring-like effect?

Correct, this is a Cartesian impedance control.

> In other words, make the joint torque track the magnitude of the torque before the external force is applied. This approach also appears to be simpler.

Sounds like a great idea! I haven't thought about this yet.

Thanks again for these great insights and detailed explanations! Is it okay to close this issue for now and solve the invalid joint names in #244?

abaicoming, 2025-01-18:
Once again, thank you for your helpful responses.

> Is it okay to close this issue for now and solve the invalid joint names in https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/244?

Of course, feel free to do so.

closed 2025-01-25 by mhubii