lbr-stack/lbr_fri_ros2_stack #243: Cartesian Overlay Support

RLi43, 2024-12-18:
Hello, do you have any plan to implement FRI Cartesian Overlay?

I can see that at least we need to
* define a hardware component "Cartesian" GPIO
* implement a switch between Joint Overlay and Cartesian Overlay, as they can't coexist
* implement an additional communication channel with the server changing the overlay type if we would like to switch it on the fly

mhubii, 2025-01-04:
hi @RLi43, apologize the late response. What you suggest is correct. I don't have access to FRI 2.x so cannot test this. However, I am happy to collaborate and walk you through the steps if you are willing to create pull request.

As a quick workaround, one way to command twists is to run

```shell
ros2 launch lbr_bringup hardware.launch.py ctrl:=twist_controller
```

RLi43, 2025-01-04:
@mhubii Thanks for your reply and happy new year : )

I did some work and implemented cartesian overlay but it's very rudimental. I'd better clean my code further before creating pull requests.
On the other hand, I started to doubt whether it makes sense to use Cartesian Overlay. My initial motivation is to have all the kinematic calculations done on the robot itself as it's faster and more precise. But they are inevitable once you start to consider the valid joint angle ranges, obstacle avoidance using the null space, shortest path from the current configuration, ...

RLi43, 2025-01-10:
I will update my changes on my forks:
https://github.com/RLi43/lbr_fri_ros2_stack/tree/lbr-stack
https://github.com/RLi43/lbr_fri_idl/tree/fri-2

Currently, it works but has not been tested whether it's compatible with JointOverlay.

mhubii, 2025-01-11:
thanks for this @RLi43, lets keep this issue open for now. Would you mind creating a pull request?

RLi43, 2025-01-11:
Sure, I'd like to, once I test the compatibility with joint overlay. 
Also I modified the command interface logic trying to avoid triggering the speed limit stop.