lbr-stack/lbr_fri_ros2_stack #244: Invalid joint names

mhubii, 2025-01-04:
>              hello @mhubii I'm having some problems replicating this demo.
It's at [https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/humble/lbr_demos/lbr_demos_py/lbr_demos_py/joint_trajectory_client.py](url)
The following error occurs when no modifications are made
```
[INFO] [1735882172.589733034] [lbr.joint_trajectory_client]: Action server available.
[INFO] [1735882172.590060335] [lbr.joint_trajectory_client]: Rotating odd joints.
[ERROR] [1735882172.591373681] [lbr.joint_trajectory_client]: Goal was rejected by server.
[INFO] [1735882172.591650790] [lbr.joint_trajectory_client]: Moving to zero position.
[ERROR] [1735882172.592469636] [lbr.joint_trajectory_client]: Goal was rejected by server.
```
At the same time, the following ERROR occurs in the lbr_bringup terminal
```[ruby $(which ign) gazebo-3] [INFO] [1735882172.590558540] [lbr.joint_trajectory_controller]: Received new action goal
[ruby $(which ign) gazebo-3] [ERROR] [1735882172.590616486] [lbr.joint_trajectory_controller]: Incoming joint A1 doesn't match the controller's joints.
[ruby $(which ign) gazebo-3] [INFO] [1735882172.591971219] [lbr.joint_trajectory_controller]: Received new action goal
[ruby $(which ign) gazebo-3] [ERROR] [1735882172.592006359] [lbr.joint_trajectory_controller]: Incoming joint A1 doesn't match the controller's joints.
```
In gazebo Entity tree, the joint name is lbr_A1.. So I changed line 38 in the file joint_trajectory_client.py from ```joint_trajectory_goal.trajectory.joint_names.append(f"A{i + 1}")```
 to
 ```joint_trajectory_goal.trajectory.joint_names.append(f"lbr_A{i + 1}")```
There are no errors in the lbr_bringup  terminal, and the action seemed to be accepted
```
[ruby $(which ign) gazebo-3] [INFO] [1735882893.824892561] [lbr.joint_trajectory_controller]: Received new action goal
[ruby $(which ign) gazebo-3] [INFO] [1735882893.824969735] [lbr.joint_trajectory_controller]: Accepted new action goal
```
BUT the following error occurs about two seconds after running again:
```
[INFO] [1735882893.823955360] [lbr.joint_trajectory_client]: Action server available.
[INFO] [1735882893.824274793] [lbr.joint_trajectory_client]: Rotating odd joints.
[INFO] [1735882893.825519684] [lbr.joint_trajectory_client]: Goal was accepted by server.
Traceback (most recent call last):
  File "/home/u051/lbr-stack/install/lbr_demos_py/lib/lbr_demos_py/joint_trajectory_client", line 33, in <module>
    sys.exit(load_entry_point('lbr-demos-py', 'console_scripts', 'joint_trajectory_client')())
  File "/home/u051/lbr-stack/build/lbr_demos_py/lbr_demos_py/joint_trajectory_client.py", line 73, in main
    joint_trajectory_client.execute(
  File "/home/u051/lbr-stack/build/lbr_demos_py/lbr_demos_py/joint_trajectory_client.py", line 60, in execute
    result_future.result().result.error_code
AttributeError: 'NoneType' object has no attribute 'result'
[ros2run]: Process exited with failure 1
```
So far I don't know how to solve it, I am indeed very beginner in ros, please forgive my unprofessional description and tell me what to do. thanks!

_Originally posted by @BabuBabuDUT in https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/162#issuecomment-2567807075_

RLi43, 2025-01-10:
I met this issue before. 
Simply it's because that the joints are named `lbr_Ai` in the [configuration file](https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_description/ros2_control/lbr_controllers.yaml#L56). And actually all the other controllers ([for example](https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_ros2_control/src/controllers/lbr_joint_position_command_controller.cpp#L85)) follow the same format.

mhubii, 2025-01-11:
for anyone stumbling across this, the demos need to be prefixed with `lbr_` (or the robot name generally speaking), here:

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_demos/lbr_demos_cpp/src/joint_trajectory_client.cpp#L48

and here:

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_demos/lbr_demos_py/lbr_demos_py/joint_trajectory_client.py#L38C46-L38C57

mhubii, 2025-03-18:
this was patched in #256 and #257 by @Khouloud-Hamrouni, thanks! Ideally, the prefix `lbr_` should be variable (as it changes with `robot_name`). This should be supported moving forward.

closed 2025-03-18 by mhubii