lbr-stack/lbr_fri_ros2_stack #245: Velocity limit error in lbr_fri_ros2 when sending rapidly changing target poses

fromlin, 2025-01-13:
Hello,  

I followed the example from **`lbr_demos_advanced_cpp - Pose Controller`** and confirmed that it works.
(https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_advanced_cpp/doc/lbr_demos_advanced_cpp.html#pose-controller)
Afterward, I wrote a new node to integrate the pose data from an Omega device (Force Dimension) to calculate and send real-time target poses for the robot.

The process works as follows:
- Define the transformation relationship between the Omega device's coordinate system and the KUKA robot's end frame, and apply this in the calculations.
- Receive `/state/pose` and `/omega/pose` → Calculate target pose → Transmit `/command/pose`
- Launch and run the following:
  1. `ros2 launch fd_bringup fd.launch.py` (Omega connection)
  2. `ros2 launch lbr_bringup hardware.launch.py ctrl:=lbr_joint_position_command_controller model:=med14` (KUKA hardware launch)
  3. Run `LBRServer.java` on the KUKA application.
  4. `ros2 run lbr_demos_advanced_cpp pose_control --ros-args -r __ns:=/lbr`
  5. `ros2 run lbr_demos_advanced_cpp pose_omega` (pose omega node)

In most cases, the robot moves as intended. However, when I move the Omega device quickly (causing the target pose to change rapidly), an error occurs in the `hardware.launch.py` process.  

Here is the error log:
```shell
[ros2_control_node-3] [ERROR] [1736749284.112445019] [lbr_fri_ros2::CommandGuard]: Velocity not in limits for joint 'lbr_A6'
[ros2_control_node-3] [ERROR] [1736749284.112475398] [lbr_fri_ros2::PositionCommandInterface]: Invalid command.
[ros2_control_node-3] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-3]   what():  Invalid command.
[ros2_control_node-3] Stack trace (most recent call last) in thread 33359:
[ros2_control_node-3] #14   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-3] #13   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x725a679ad84f]
[ros2_control_node-3] #12   Source "./nptl/pthread_create.c", line 442, in start_thread [0x725a6791bac2]
[ros2_control_node-3] #11   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x725a67bac252, in 
[ros2_control_node-3] #10   Object "/home/mras/ws/ros2_humble/kuka/install/lbr_fri_ros2/lib/liblbr_fri_ros2.so", at 0x725a606a24af, in lbr_fri_ros2::Worker::run_async(int)::{lambda()#1}::operator()() const
[ros2_control_node-3] #9    Object "/home/mras/ws/ros2_humble/kuka/install/lbr_fri_ros2/lib/liblbr_fri_ros2.so", at 0x725a60667413, in lbr_fri_ros2::App::perform_work_()
[ros2_control_node-3] #8    Object "/home/mras/ws/ros2_humble/kuka/install/fri_client_sdk/lib/libFRIClient.so", at 0x725a6053a776, in KUKA::FRI::ClientApplication::step()
[ros2_control_node-3] #7    Object "/home/mras/ws/ros2_humble/kuka/install/lbr_fri_ros2/lib/liblbr_fri_ros2.so", at 0x725a6065972c, in lbr_fri_ros2::PositionCommandInterface::buffered_command_to_fri(KUKA::FRI::LBRCommand&, lbr_fri_idl::msg::LBRState_<std::allocator<void> > const&) [clone .cold]
[ros2_control_node-3] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x725a67b7e4d7, in __cxa_throw
[ros2_control_node-3] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x725a67b7e276, in std::terminate()
[ros2_control_node-3] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x725a67b7e20b, in 
[ros2_control_node-3] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x725a67b72b9d, in 
[ros2_control_node-3] #2    Source "./stdlib/abort.c", line 79, in abort [0x725a678af7f2]
[ros2_control_node-3] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x725a678c9475]
[ros2_control_node-3] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-3]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-3]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x725a6791d9fc]
[ros2_control_node-3] Aborted (Signal sent by tkill() 33322 1000)
[ERROR] [ros2_control_node-3]: process has died [pid 33322, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/lbr --params-file /tmp/launch_params_rkom8gp9 --params-file /home/mras/ws/ros2_humble/kuka/install/lbr_description/share/lbr_description/ros2_control/lbr_controllers.yaml --params-file /tmp/launch_params_2q2p4b62 -r ~/robot_description:=robot_description'].
``` 

It seems the `lbr_A6` joint, which corresponds to one of the joints requiring significant movement to accommodate the target pose, exceeds its velocity limits when the target poses change too rapidly.  

### **Question**:
1. Are there any modifications or configurations within the **`lbr_fri_ros2_stack`** that can address this issue?  
2. Could you recommend any references or materials that might help resolve this problem?

Thank you for your assistance.

fromlin, 2025-01-13:
Thank you for your response @mhubii .

I forgot to mention earlier that I had already reviewed the file you suggested (**`med14/joint_limits.yaml`**, which corresponds to the robot model I am using), and I agree that it is best not to modify the parameters recommended by KUKA. (Personally, I also feel that the default settings are fast enough.)

However, I tried something else in **`LBRServer.java`**. 

Specifically, I added the following line of code:

```java
fri_overlay_.overrideJointAcceleration(15.0);   // Range: 1.0 ~ 20.0
```

This adjustment allowed the system to withstand higher velocities without encountering the error as frequently. However, it didn’t eliminate the errors entirely.

I haven't tried increasing the value to 20 yet, but I believe it wouldn't be a fundamental solution to the problem.

mhubii, 2025-01-13:
alternatively, in this code (not implemented by me):

https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/b615b9caf0b7420d7269fa5027c0c70daabf40b3/lbr_demos/lbr_demos_advanced_cpp/src/pose_control_node.cpp#L74

one might add a `joint_position_target`. Given the current `joint_position`, the command could be clipped.

TalentLeshen-DLUT, 2025-03-18:
@fromlin Hello, may I ask if you have implemented other solutions, because I also encountered the same problem.

fromlin, 2025-03-19:
Sorry for the delayed response.
I was only briefly testing this as part of a functional test and have not proceeded further due to other tasks.  

I attempted the method suggested by @mhubii to clip `joint_position_command` to its limits by adding code:

```
#include "geometry_msgs/msg/pose_stamped.hpp"
#include "lbr_fri_ros2/command_guard.hpp"

...

class PoseControlNode : public LBRBasePositionCommandNode
{
protected:
  
  ...

  rclcpp::Publisher<geometry_msgs::msg::PoseStamped>::SharedPtr joint_pub_;
  lbr_fri_ros2::CommandGuardParameters cmd_guard_params_;
  geometry_msgs::msg::PoseStamped joint_velocity_;

public:
  PoseControlNode(const rclcpp::NodeOptions &options)
      : LBRBasePositionCommandNode("pose_control", options)
  {
    KDL::Tree robot_tree;
    if (!kdl_parser::treeFromString(this->robot_description_, robot_tree))
    {
      RCLCPP_ERROR(this->get_logger(), "Failed to construct KDL tree.");
      return;
    }

    this->declare_parameter<std::string>("base_link", "lbr_link_0");
    this->declare_parameter<std::string>("end_effector_link", "lbr_link_ee");

    std::string root_link = this->get_parameter("base_link").as_string();
    std::string tip_link = this->get_parameter("end_effector_link").as_string();

    if (!robot_tree.getChain(root_link, tip_link, chain_))
    {
      RCLCPP_ERROR(this->get_logger(), "Failed to get chain from tree.");
    }
    else
    {
      RCLCPP_INFO(this->get_logger(), "Successfully retrieved chain from tree.");
    }

    /* ---------------------------- Added command_guard to MED14 parameters ---------------------------- */
    double tmp_scale = 5.0;
    cmd_guard_params_.joint_names = {"lbr_A1", "lbr_A2", "lbr_A3", "lbr_A4",
                                     "lbr_A5", "lbr_A6", "lbr_A7"};
    cmd_guard_params_.max_positions = {170. * M_PI / 180., 120. * M_PI / 180., 170. * M_PI / 180.,
                                       120. * M_PI / 180., 170. * M_PI / 180., 120. * M_PI / 180.,
                                       175. * M_PI / 180.};
    cmd_guard_params_.min_positions = {-170. * M_PI / 180., -120. * M_PI / 180., -170. * M_PI / 180.,
                                       -120. * M_PI / 180., -170. * M_PI / 180., -120. * M_PI / 180.,
                                       -175. * M_PI / 180.};
    cmd_guard_params_.max_velocities = {85. * M_PI / 180. * tmp_scale, 85. * M_PI / 180. * tmp_scale, 100. * M_PI / 180. * tmp_scale,
                                        75. * M_PI / 180. * tmp_scale, 130. * M_PI / 180. * tmp_scale, 135. * M_PI / 180. * tmp_scale,
                                        135. * M_PI / 180. * tmp_scale};
    cmd_guard_params_.max_torques = {200., 200., 200., 200., 200., 200., 200.};

    joint_pub_ = this->create_publisher<geometry_msgs::msg::PoseStamped>("command/joint_velocity", 1);
    /* ------------------------------------------------------------------------------------------------ */

    pose_pub_ = this->create_publisher<geometry_msgs::msg::Pose>("state/pose", 1);
    pose_sub_ = this->create_subscription<geometry_msgs::msg::Pose>(
        "command/pose", 1, std::bind(&PoseControlNode::on_pose_, this, std::placeholders::_1));
  }
    
[... 42 lines cut ...]
        }

        // Position check
        joint_position_command.joint_position[i] =
        std::min(cmd_guard_params_.max_positions[i], std::max(cmd_guard_params_.min_positions[i], joint_position_command.joint_position[i]));
      }
      joint_pub_->publish(joint_velocity_);
      /* -------------------------------------------------------------------------- */

      lbr_joint_position_command_pub_->publish(joint_position_command);
    }
  }
};
```

- Included `"lbr_fri_ros2/command_guard.hpp"` to compare against limits.
- Used `"geometry_msgs::msg::PoseStamped"` to simply display the modified target joint velocities.

When using the predefined `max_velocities` directly, the motion was too slow. Upon checking the state data using **PlotJuggler**, I found a significant discrepancy between `joint_position_command.joint_position` and the actual joint positions reported in `LBRState.msg` such as measured_joint_position, commanded_joint_position.  

I had to scale up the velocity limits beyond the values defined in the config to achieve reasonable movement speed (though I know this is not the correct solution). However, as I mentioned at the beginning, I have moved on to other tasks and have not investigated further. 

If you want to resolve this issue, I recommend checking the `LBRState` data and comparing the commanded values with what the robot actually receives.  

I sincerely hope you can find a proper solution. If you manage to fix it, I would really appreciate it if you could share your findings :)