lbr-stack/lbr_fri_ros2_stack #250: How to use force control with ros.

josefinemonnet, 2025-02-23:
I am currently controlling the robot with the moveit api using position control. How is it possible to overlay a certain force value in z-direction?

mhubii, 2025-02-26:
hi @josefinemonnet, this currently requires enhancements, also refers to https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/115. Apologies.

You can run the wrench command controller stand-alone:

https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_cpp/doc/lbr_demos_cpp.html#lbr-wrench-command-controller-hardware-only

However, load data calibration is undocumented https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/235. I wish we had more resources for quicker improvements.

jomon250, 2025-03-04:
Thank you @mhubii for the quick_response. Just for my understanding, without adapting the currently available controllers (joint poisition and wrench), I can only control either the wrench or the joint positions, right? So what I have to do is to write a controller which takes wrench as well as joint position commands and sends them to the interface? [Here](https://lbr-stack.readthedocs.io/en/latest/lbr_fri_idl.html#command-interfaces) it says that with the LBRWrenchCommand.msg I may command wrench as well as joints, right?

jomon250, 2025-03-05:
Another opportunity would be to simply use the Cartesian Impedance Controller, wouldn't it? I could tune the stiffness-value in z accordingly, and the other stiffness values relatively high, so that it follows the commanded trajectory. I am just wondering why it says [here](https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_bringup/doc/lbr_bringup.html), that I have to choose between the position and the joint impedance controller but not the cartesian impedance controller?