lbr-stack/lbr_fri_ros2_stack #251: Awaiting 'COMMANDING_WAIT' state.

ZixuanLiu-Justin, 2025-02-25:
Hi,

I'm trying to run the [LBR Joint Position Command Controller (Hardware only)](https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_cpp/doc/lbr_demos_cpp.html) on the KUKA Med14. I've set the robot to automatic mode (AUT). From the output on the smartPad, it shows the connection actually established, but then it immediately closed itself.

[image]

below is the output from the terminal, it seems like the robot is never going to the expected state 'COMMANDING_WAIT'.

`
~/rosworkspace$ ros2 launch lbr_bringup hardware.launch.py ctrl:=lbr_joint_position_command_controller model:=med14
[INFO] [launch]: All log files can be found below /home/zixuan/.ros/log/2025-02-25-16-37-06-100114-zixuan-bmlubuntu-9480
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [9483]
[INFO] [robot_state_publisher-2]: process started with pid [9485]
[INFO] [ros2_control_node-3]: process started with pid [9487]
[INFO] [spawner-4]: process started with pid [9489]
[INFO] [spawner-5]: process started with pid [9491]
[INFO] [spawner-6]: process started with pid [9493]
[INFO] [spawner-7]: process started with pid [9495]
[static_transform_publisher-1] [INFO] [1740523026.260632058] [static_transform_publisher_lkUzBQP5Zh3wIVxy]: Spinning until stopped - publishing transform
[static_transform_publisher-1] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-1] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-1] from 'world' to 'lbr_floating_link'
[robot_state_publisher-2] [INFO] [1740523026.262376564] [lbr.robot_state_publisher]: got segment lbr_floating_link
[robot_state_publisher-2] [INFO] [1740523026.262467866] [lbr.robot_state_publisher]: got segment lbr_link_0
[robot_state_publisher-2] [INFO] [1740523026.262471443] [lbr.robot_state_publisher]: got segment lbr_link_1
[robot_state_publisher-2] [INFO] [1740523026.262473707] [lbr.robot_state_publisher]: got segment lbr_link_2
[robot_state_publisher-2] [INFO] [1740523026.262475530] [lbr.robot_state_publisher]: got segment lbr_link_3
[robot_state_publisher-2] [INFO] [1740523026.262477364] [lbr.robot_state_publisher]: got segment lbr_link_4
[robot_state_publisher-2] [INFO] [1740523026.262479177] [lbr.robot_state_publisher]: got segment lbr_link_5
[robot_state_publisher-2] [INFO] [1740523026.262480920] [lbr.robot_state_publisher]: got segment lbr_link_6
[robot_state_publisher-2] [INFO] [1740523026.262482634] [lbr.robot_state_publisher]: got segment lbr_link_7
[robot_state_publisher-2] [INFO] [1740523026.262484277] [lbr.robot_state_publisher]: got segment lbr_link_ee
[ros2_control_node-3] [WARN] [1740523026.266121267] [lbr.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1740523026.266279835] [resource_manager]: Loading hardware 'lbr_system_interface' 
[ros2_control_node-3] [INFO] [1740523026.268099392] [resource_manager]: Initialize hardware 'lbr_system_interface' 
[ros2_control_node-3] [INFO] [1740523026.268136872] [lbr_fri_ros2::AsyncClient]: Configuring client
[ros2_control_node-3] [INFO] [1740523026.268142142] [lbr_fri_ros2::AsyncClient]: Client command mode: 'JOINT_POSITION'
[ros2_control_node-3] [INFO] [1740523026.268144246] [lbr_fri_ros2::AsyncClient]: Command guard variant 'default'
[ros2_control_node-3] [INFO] [1740523026.268153333] [lbr_fri_ros2::CommandGuard]: *** Parameters:
[ros2_control_node-3] [INFO] [1740523026.268155697] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A1 limits: Position: [-170.0, 170.0] deg, velocity: 85.0 deg/s, torque: 200.0 Nm
[ros2_control_node-3] [INFO] [1740523026.268160096] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A2 limits: Position: [-120.0, 120.0] deg, velocity: 85.0 deg/s, torque: 200.0 Nm
[ros2_control_node-3] [INFO] [1740523026.268162660] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A3 limits: Position: [-170.0, 170.0] deg, velocity: 100.0 deg/s, torque: 200.0 Nm
[ros2_control_node-3] [INFO] [1740523026.268164504] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A4 limits: Position: [-120.0, 120.0] deg, velocity: 75.0 deg/s, torque: 200.0 Nm
[ros2_control_node-3] [INFO] [1740523026.268166167] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A5 limits: Position: [-170.0, 170.0] deg, velocity: 130.0 deg/s, torque: 200.0 Nm
[ros2_control_node-3] [INFO] [1740523026.268167810] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A6 limits: Position: [-120.0, 120.0] deg, velocity: 135.0 deg/s, torque: 200.0 Nm
[ros2_control_node-3] [INFO] [1740523026.268169503] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A7 limits: Position: [-175.0, 175.0] deg, velocity: 135.0 deg/s, torque: 200.0 Nm
[ros2_control_node-3] [INFO] [1740523026.268172659] [lbr_fri_ros2::JointExponentialFilterArray]: *** Parameters:
[ros2_control_node-3] [INFO] [1740523026.268174372] [lbr_fri_ros2::JointExponentialFilterArray]: *   tau: 0.04000 s
[ros2_control_node-3] [INFO] [1740523026.268178440] [lbr_fri_ros2::StateInterface]: *** Parameters:
[ros2_control_node-3] [INFO] [1740523026.268179742] [lbr_fri_ros2::StateInterface]: *   external_torque_tau: 0.04000 s
[ros2_control_node-3] [INFO] [1740523026.268181325] [lbr_fri_ros2::StateInterface]: *   measured_torque_tau: 0.04000 s
[ros2_control_node-3] [INFO] [1740523026.268183309] [lbr_fri_ros2::AsyncClient]: Open loop 'true'
[ros2_control_node-3] [INFO] [1740523026.268184932] [lbr_fri_ros2::AsyncClient]: Client configured
[ros2_control_node-3] [INFO] [1740523026.269129195] [resource_manager]: Successful initialization of hardware 'lbr_system_interface'
[ros2_control_node-3] [INFO] [1740523026.269191753] [resource_manager]: 'configure' hardware 'lbr_system_interface' 
[ros2_control_node-3] [INFO] [1740523026.269195290] [resource_manager]: Successful 'configure' of hardware 'lbr_system_interface'
[ros2_control_node-3] [INFO] [1740523026.269202313] [resource_manager]: 'activate' hardware 'lbr_system_interface' 
[ros2_control_node-3] [INFO] [1740523026.269208645] [lbr_fri_ros2::App]: Opening UDP socket with port_id '30200'
[ros2_control_node-3] [INFO] [1740523026.269219024] [lbr_fri_ros2::App]: Socket opened successfully
[ros2_control_node-3] [INFO] [1740523026.269255453] [lbr_ros2_control::SystemInterface]: Awaiting robot heartbeat. Attempt 1, remote_host 'INADDR_ANY', port_id '30200'
[ros2_control_node-3] [WARN] [1740523026.269283215] [lbr_fri_ros2::App]: Failed to set FIFO realtime scheduling policy. Refer to [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html].
[ros2_control_node-3] [INFO] [1740523026.269297822] [lbr_fri_ros2::App]: Starting run thread
[spawner-4] [INFO] [1740523026.427681946] [lbr.spawner_joint_state_broadcaster]: waiting for service /lbr/controller_manager/list_controllers to become available...
[spawner-6] [INFO] [1740523026.427719707] [lbr.spawner_lbr_state_broadcaster]: waiting for service /lbr/controller_manager/list_controllers to become available...
[spawner-5] [INFO] [1740523026.427856183] [lbr.spawner_force_torque_broadcaster]: waiting for service /lbr/controller_manager/list_controllers to become available...
[spawner-7] [INFO] [1740523026.436675716] [lbr.spawner_lbr_joint_position_command_controller]: waiting for service /lbr/controller_manager/list_controllers to become available...
[ros2_control_node-3] [INFO] [1740523027.269368842] [lbr_ros2_control::SystemInterface]: Awaiting robot heartbeat. Attempt 2, remote_host 'INADDR_ANY', port_id '30200'
[ros2_control_node-3] [INFO] [1740523028.269850677] [lbr_ros2_control::SystemInterface]: Awaiting robot heartbeat. Attempt 3, remote_host 'INADDR_ANY', port_id '30200'
[ros2_control_node-3] [INFO] [1740523029.270225436] [lbr_ros2_control::SystemInterface]: Awaiting robot heartbeat. Attempt 4, remote_host 'INADDR_ANY', port_id '30200'
[... 26 lines cut ...]
[ros2_control_node-3] [INFO] [1740523041.275472275] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'IDLE'.
[ros2_control_node-3] [INFO] [1740523042.276457101] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'IDLE'.
[ros2_control_node-3] [INFO] [1740523043.276959485] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'IDLE'.
[... cut ...]
[ros2_control_node-3] [INFO] [1740523048.279158573] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'IDLE'.
[ros2_control_node-3] [INFO] [1740523049.279660122] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'IDLE'.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
`

I'm using the ros2 humble with naive Ubuntu 22.04, the KUKA comes with fri version 2.5. Any ideas can potentially solve this?

BTW, I'm also attaching a photo showing a safety warning message here:

[image]

mhubii, 2025-02-26:
hi @ZixuanLiu-Justin, thanks for providing these detailed reports. For some reason your robot disconnects immediately:

```shell
[ros2_control_node-3] [INFO] [1740523033.929589624] [lbr_fri_ros2::AsyncClient]: LBR switched from 'MONITORING_WAIT' to 'MONITORING_READY'
[ros2_control_node-3] [INFO] [1740523034.002722950] [lbr_fri_ros2::AsyncClient]: LBR switched from 'MONITORING_READY' to 'IDLE'
```

The safety warnings are fine. I am assuming you installed this Java application: https://github.com/lbr-stack/fri/blob/fri-2.5/server_app/LBRServer.java

@fredRocs could you please give me a helping hand here (I recall you used FRI `2.5`?)?

ZixuanLiu-Justin, 2025-02-26:
Hi @mhubii, thank you for the information!

I can confirm that I successfully connect to the robot by using:

```
ros2 launch lbr_bringup hardware.launch.py model:=med14
```
and I can control the robot by using moveit via:
```
ros2 launch lbr_bringup move_group.launch.py mode:=mock rviz:=true model:=med14
```
I tested in `T1` mode and then switched to `AUT`, it still works.

It currently should work fine for me for my need:)

mhubii, 2025-02-26:
Okay very glad to hear, thank you very much for confirming! :) If you find some insights on why it didn't work, it would be very valuable for the community to share them. I do not have access to Sunrise Workbench with 2.5 and can thus not always verify the server side (robot side).

garylvov, 2025-09-18:
Hi, I have the same warnings.

Mine also doesn't work in the same way; I suspect this may be because of mismatching FRI versions on robot / in the workspace, but I'm not sure, I will try to confirm soon.

[image]

[image]

```
dezire@dezire:~/lbr-stack$ ros2 launch lbr_bringup hardware.launch.py
[INFO] [launch]: All log files can be found below /home/dezire/.ros/log/2025-09-18-17-35-21-779275-dezire-45318
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [45321]
[INFO] [ros2_control_node-2]: process started with pid [45323]
[INFO] [spawner-3]: process started with pid [45325]
[INFO] [spawner-4]: process started with pid [45327]
[INFO] [spawner-5]: process started with pid [45329]
[INFO] [spawner-6]: process started with pid [45331]
[robot_state_publisher-1] [INFO] [1758231321.996626977] [lbr.robot_state_publisher]: got segment lbr_link_0
[robot_state_publisher-1] [INFO] [1758231321.996685013] [lbr.robot_state_publisher]: got segment lbr_link_1
[robot_state_publisher-1] [INFO] [1758231321.996690371] [lbr.robot_state_publisher]: got segment lbr_link_2
[robot_state_publisher-1] [INFO] [1758231321.996693226] [lbr.robot_state_publisher]: got segment lbr_link_3
[robot_state_publisher-1] [INFO] [1758231321.996696192] [lbr.robot_state_publisher]: got segment lbr_link_4
[robot_state_publisher-1] [INFO] [1758231321.996698598] [lbr.robot_state_publisher]: got segment lbr_link_5
[robot_state_publisher-1] [INFO] [1758231321.996701049] [lbr.robot_state_publisher]: got segment lbr_link_6
[robot_state_publisher-1] [INFO] [1758231321.996703332] [lbr.robot_state_publisher]: got segment lbr_link_7
[robot_state_publisher-1] [INFO] [1758231321.996705545] [lbr.robot_state_publisher]: got segment lbr_link_ee
[robot_state_publisher-1] [INFO] [1758231321.996707905] [lbr.robot_state_publisher]: got segment world
[ros2_control_node-2] [WARN] [1758231322.001190304] [lbr.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-2] [INFO] [1758231322.001420307] [resource_manager]: Loading hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1758231322.003461858] [resource_manager]: Initialize hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1758231322.003513343] [lbr_fri_ros2::AsyncClient]: Configuring client
[ros2_control_node-2] [INFO] [1758231322.003519426] [lbr_fri_ros2::AsyncClient]: Client command mode: 'POSITION'
[ros2_control_node-2] [INFO] [1758231322.003521632] [lbr_fri_ros2::AsyncClient]: Command guard variant 'default'
[ros2_control_node-2] [INFO] [1758231322.003534240] [lbr_fri_ros2::CommandGuard]: *** Parameters:
[ros2_control_node-2] [INFO] [1758231322.003536721] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A1 limits: Position: [-170.0, 170.0] deg, velocity: 98.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1758231322.003542698] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A2 limits: Position: [-120.0, 120.0] deg, velocity: 98.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1758231322.003545618] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A3 limits: Position: [-170.0, 170.0] deg, velocity: 100.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1758231322.003548205] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A4 limits: Position: [-120.0, 120.0] deg, velocity: 130.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1758231322.003550834] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A5 limits: Position: [-170.0, 170.0] deg, velocity: 140.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1758231322.003553500] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A6 limits: Position: [-120.0, 120.0] deg, velocity: 180.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1758231322.003556128] [lbr_fri_ros2::CommandGuard]: *   Joint lbr_A7 limits: Position: [-175.0, 175.0] deg, velocity: 180.0 deg/s, torque: 200.0 Nm
[ros2_control_node-2] [INFO] [1758231322.003559838] [lbr_fri_ros2::JointExponentialFilterArray]: *** Parameters:
[ros2_control_node-2] [INFO] [1758231322.003561795] [lbr_fri_ros2::JointExponentialFilterArray]: *   tau: 0.04000 s
[ros2_control_node-2] [INFO] [1758231322.003566830] [lbr_fri_ros2::StateInterface]: *** Parameters:
[ros2_control_node-2] [INFO] [1758231322.003568554] [lbr_fri_ros2::StateInterface]: *   external_torque_tau: 0.04000 s
[ros2_control_node-2] [INFO] [1758231322.003570613] [lbr_fri_ros2::StateInterface]: *   measured_torque_tau: 0.04000 s
[ros2_control_node-2] [INFO] [1758231322.003572883] [lbr_fri_ros2::AsyncClient]: Open loop 'true'
[ros2_control_node-2] [INFO] [1758231322.003574960] [lbr_fri_ros2::AsyncClient]: Client configured
[ros2_control_node-2] [INFO] [1758231322.004658320] [resource_manager]: Successful initialization of hardware 'lbr_system_interface'
[ros2_control_node-2] [INFO] [1758231322.004733257] [resource_manager]: 'configure' hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1758231322.004738372] [resource_manager]: Successful 'configure' of hardware 'lbr_system_interface'
[ros2_control_node-2] [INFO] [1758231322.004747678] [resource_manager]: 'activate' hardware 'lbr_system_interface' 
[ros2_control_node-2] [INFO] [1758231322.004754177] [lbr_fri_ros2::App]: Opening UDP socket with port_id '30200'
[... cut ...]
[spawner-3] [INFO] [1758231332.196593772] [lbr.spawner_joint_state_broadcaster]: waiting for service /lbr/controller_manager/list_controllers to become available...
[ros2_control_node-2] [INFO] [1758231333.007243850] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'IDLE'.
[ros2_control_node-2] [INFO] [1758231334.007461571] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'IDLE'.
[ros2_control_node-2] [INFO] [1758231335.007620158] [lbr_ros2_control::SystemInterface]: Awaiting 'COMMANDING_WAIT' state. Current state 'IDLE'.
 ```

mhubii, 2025-09-19:
hi @garylvov , thanks for the detailed feedback. It seems that in your case the client side (your laptop) doesn't keep up with the 1KHz communication rate. Thus the server side (KUKA controller) shuts down the connection

garylvov, 2025-09-19:
I've used this laptop with other kukas successfully before - didn't work with the new kuka even with 10ms send period. I will try some things later today and get back to you thank you!

mhubii, 2025-09-19:
I see, but the latency seems to be the issue here. Sometimes people report using another Ethernet cable helps.

Otherwise I'd suggest running KUKA's example applications standalone. If they work fine, then the root issue if this stack.