lbr-stack/lbr_fri_ros2_stack #253: Building the entire stack for Ros 2 Humble Fails

Srinidhi-R, 2025-03-16:
I tried to build the package on the system with below details:

OS: Ubuntu 22.04
Ros version: Humble

Whenever the colcon build step was executed, the build gets stuck at the lbr_demos_cpp package, and the terminal closes.

I checked the events.log and logger_all.log files and couldn't find the error point. I built each package individually with the --executor-sequential option and all packages build successfully except the lbr_demos_cpp, which gives the same error and closes the terminal.

Does this package have any prerequisites that are needed? Is there any way to fix this?

mhubii, 2025-03-18:
hi @Srinidhi-R, this is quite strange indeed. This software gets compiled routinely through GItHub actions and works well for `humble`.

Could this be some RAM limitation on your system? I see you already tried `--executor sequential`, could you also do `--parallel-workers 1`?

If indeed this is an issue with `lbr_demos_cpp`, you can try `--packages-ignore lbr_demos_cpp`. Let me know if any of this helps.

Srinidhi-R, 2025-03-18:
It seems to be an issue with multiple packages including lbr_ros2_control.
I have tried parallel workers option also, with no success.

TalentLeshen-DLUT, 2025-03-18:
A simple method is first to remove `lbr_moveit_cpp` for `colcon build`, and then add `lbr_moveit_cpp` for `colcon build`. That would solve the problem, but I don't know why.

mhubii, 2025-03-18:
To rule out system-dependent issues, you can try run inside a clean Docker container: https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/docker/doc/docker.html

mhubii, 2025-03-26:
any update here @Srinidhi-R?

mhubii, 2025-03-27:
okay, no worries! Please do let me know how it goes when you find time.