lbr-stack/lbr_fri_ros2_stack #260: Robot does not move in Joint Trajectory Controller simulation

Khouloud-Hamrouni, 2025-03-24:
Hello,
I'm encountering an issue when following the [quick start guide](https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_demos_cpp/doc/lbr_demos_cpp.html#simulation) when trying to run the joint_trajectory_client. 
When running the simulation, everything works correctly, the robot appears on Gazebo and it shows that the goal was accepted by server. 
But, the robot does not move to the target pose set in hello_moveit.cpp. 
I tried changing the target pose but nothing happened.
All I need is a working example to know how to command my robot on Gazebo simulation.
I would also appreciate it If there is any other working examples for motion planning on Moveit that's applied on Gazebo simulation.

Environment:
OS: **Ubuntu-22.04**
ROS2 Distro: **Humble**
FRI version: **2.2.0**

Thanks in advance for your time!

mhubii, 2025-03-25:
Hi @Khouloud-Hamrouni. I think what you may be encountering here is a slow real time factor, see bottom right simulation:

[image]

You can modify the integration step-size in `/usr/share/ignition/ignition-gazebo6/worlds/empty.sdf`:

```xml
<world name="empty">
  <physics name="1ms" type="ignored">
    <max_step_size>0.001</max_step_size>
```

to `<max_step_size>0.01</max_step_size>`.

However, this is likely to do with the high resolution of the collision meshes. The resolution of the collision meshes, e.g. https://github.com/lbr-stack/lbr_fri_ros2_stack/tree/humble/lbr_description/meshes/iiwa7/collision, should be reduced in this repository.

mhubii, 2025-03-25:
thank you for confirming shortly