lbr-stack/lbr_fri_ros2_stack #272: KONI IP Address for FRI Installation

Khouloud-Hamrouni, 2025-04-04:
Hello! This is more of a question than an issue, but I thought someone here might have encountered it before and could guide me through it.

I followed the explications [here](https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_fri_ros2_stack/doc/hardware_setup.html#install-applications-to-the-robot) and I tried to install the lbr_fri_ros2 application on my KUKA robot (LBR IIWA 14, OS 1.16.0.7).

In StationSetup.cat, when I select the FRI-related packages (FRI Extension and Example Applications), it asks for the KONI IP address, but: 

- I don't connect my PC to KONI, only via X66 over Ethernet.
- I don't know the KONI IP address of the robot and cannot find it.

If I install an application without selecting FRI packages, the installation works fine.
But when I try to install using the default KONI IP (192.170.10.2), I get an installation error: 
*" Le chemin de réseau n'a pas été trouvé " in the issue means that the network path cannot be found.*

[image]

Questions:

- Do I need to connect my PC to KONI for FRI to work?
- How can I find the KONI IP address of my robot? It's not mentioned in the SmartPad.
- If KONI is not required, what should I enter in the Kuka Option Network Interface IP field in StationSetup.cat?

Thanks in advance for any help!

mhubii, 2025-04-04:
To install applications you connect to X66. FRI works over both KONI and X66. In this configuration in `StationSetup.cat` -> `Configuration`:

https://lbr-stack.readthedocs.io/en/latest/_images/02_station_setup_configuration1.png

You can configure the IP address for both network interfaces X66 and KONI.

mhubii, 2025-04-15:
did you manage to resolve this problem?

Khouloud-Hamrouni, 2025-04-16:
Hi, sorry for my late response I've been busy with other work.

Yes I resolved the problem by changing the PC i'm using to open sunrise workbench and install applications.
I'm still not sure about the reason of the errors because I literally did the same thing in both PCs. 
I'll let you updated if i ever discover the real reason behind these errors.
We can close the issue for now.

Thanks!

mhubii, 2025-04-17:
no worries at all. Happy to hear it was resolved. Thank you for confirming briefly! Feel free to open issues anytime.

closed 2025-04-17 by mhubii