lbr-stack/lbr_fri_ros2_stack #274: ign_ros2_control plugin not loading in Gazebo

adalmases-dev, 2025-05-13:
Hi,

I'm currently running simulations with Gazebo Sim, version 6.17.0 (Ignition), and encountered an issue with the ROS 2 control plugin defined in the `lbr_gazebo.xacro`.

By default, the plugin is defined as: 
``` 
       <gazebo>
            <plugin name="ign_ros2_control::IgnitionROS2ControlPlugin"
                filename="ign_ros2_control-system">
                ...

```
I was able to fix it by changing the filename to include the lib prefix and .so suffix explicitly:
```        
        <gazebo>
            <plugin name="ign_ros2_control::IgnitionROS2ControlPlugin"
                filename="libign_ros2_control-system.so">
                ...
```

Am I the only one facing this issue? Can this be a compatibility problem or some misconfiguration on my end? 

A possible solution I can think of, is making the plugin filename to be parameterized in the xacro via launch arguments. If this is something necessary that is of interest to be improved, I’d be happy to submit a PR to make it parameterizable or simply adjust the default.

mhubii, 2025-05-15:
hi @adalmases-dev, thank you for sharing this finding. Apologize the late response. May I ask what version of ROS you are using?

adalmases-dev, 2025-06-22:
Hi @mhubii,
Apologies for my very delayed response—I'm the one who's late this time. You were absolutely right: the issue was on my end. As a friend of mine says, “the problem was between the keyboard and the chair.” 😅

Everything works as expected with the original setup, so I'm closing this issue.

Thanks again for the great work on this stack!

closed 2025-06-22 by adalmases-dev