lbr-stack/lbr_fri_ros2_stack #275: Failed to fetch current robot state

TalentLeshen-DLUT, 2025-05-14:
When I run
`ros2 launch lbr_bringup gazebo.launch.py model:=iiwa14 moveit:=true`
`ros2 launch lbr_bringup move_group.launch.py model:=iiwa14 mode:=gazebo rviz:=true`
After that
I added `getCurrentState()` in `hello_moveit.cpp` and ran `ros2 launch lbr_moveit_cpp hello_moveit.launch.py model:=iiwa14 mode:=gazebo`. The following error will occur:
```
Listening to joint states on topic 'joint_states'
[hello_moveit-1] [WARN] [1747187454.478512164] [moveit_ros.current_state_monitor]: No state update received within 100ms of system clock. Have been waiting for 0.981000s, timeout is 1.000000s
[hello_moveit-1] [INFO] [1747187454.478539442] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds Requested time 63.529000 but latest received state has time 0.000000.
[hello_moveit-1] Check clock synchronization if your are running ROS across multiple machines!
[hello_moveit-1] [ERROR] [1747187454.478554689] [move_group_interface]: Failed to fetch current robot state
```
I tried to directly change `use_sim_time` to True, but this didn't solve the problem. I wonder if you could give me some suggestions so that I can solve it.

TalentLeshen-DLUT, 2025-05-14:
It seems that I have solved this problem. Just remap `/joint_states` to` /lbr/joint_states` in the launch of `move_group` and `hello_moveit`. Then, in `hello_moveit.cpp`,` /joint_states `is also remapped to` /lbr/joint_states`

mhubii, 2025-05-15:
thanks for sharing and confirming the solution, this seems to be missing in the documentation: https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_demos/lbr_moveit_cpp/doc/lbr_moveit_cpp.html#moveit-via-c-api

Enzo-let, 2025-06-28:
Hello I have the same problem, and i remap /joint_states to /lbr/joint_states as you said but it sill didn't work, so if you can past the change, appreciate you