lbr-stack/lbr_fri_ros2_stack #276: measured joint position cannot be updated in real time

BoWangFromMars, 2025-05-23:
Hello, I met a fatal problem when subscribing the real-time joint position of the robot. 
I ran the command "ros2 launch lbr_bringup lbr_bringup.launch.py sim:=false ctrl:=lbr_joint_position_command_controller model:=iiwa7" and set the robot operating in Joint Impedance Control mode.  I pushed the robot to change the joint position and  subscribed the real-time joint position using "ros2 topic echo /lbr/state" at the same time. I found the 'measured torque' updated in real-time, but the 'measured joint position' was always not changed, as shown in the following picture.

[image]

I also checked the FRI Manual, the measured position should be updated in real-time, as same as the measured torque. Please see the screenshot on FRI manual as follows.

[image]

I am not sure whether something goes wrong in my robot or ? Thank you!

BoWangFromMars, 2025-05-28:
In addition, for admittance control in "lbr_demos_advanced_cpp", when I try to drag the robot  manually under this control mode, the robot moves not smoothly. I checked the code and found the external force calculated based on external torque provides by FRI message, the external torque is estimated (to my best knowledge, the manufacturer uses state observer, maybe I am wrong), but the fact is the external torque is not accurate. As a result of this, the external force on the end-effector calculated by external torque is not accurate, which makes the inside position controller in admittance controller move to the wrong direction. So when we drag it manually, the robot moves not smoothly. 
I suggest that using force/torque sensor to mount on the end-effector, and then use admittance controller. Another option is set the robot work in Impedance Control Mode (Cartesian or Joint), and send the current measured joint position to the robot in real-time. In this way, we can drag the robot in a very smooth way. I have tested this method.

mhubii, 2025-05-31:
hi @BoWangFromMars , sorry for the late response. Yes the `lbr_demos_advanced_cpp` is unfortunatly not well supported right now. The simples fix to make the robot move smoothly is to adjust `lbr_system_config.yaml`, e.g.

```yaml
# these parameters are read by the lbr_system_interface.xacro and configure the lbr_ros2_control::SystemInterface
hardware:
  fri_client_sdk: # the fri_client_sdk version is used to create the correct state interfaces lbr_system_interface.xacro
    major_version: 1
    minor_version: 15
  client_command_mode: position # the command mode specifies the user-sent commands. Available: [position, torque, wrench]
  port_id: 30200 # port id for the UDP communication. Useful in multi-robot setups
  remote_host: INADDR_ANY # the expected robot IP address. INADDR_ANY will accept any incoming connection
  rt_prio: 80 # real-time priority for the control loop
  # exponential moving average time constant for joint position commands [s]:
  #   Set tau > robot sample time for more smoothing on the commands
  #   Set tau << robot sample time for no smoothing on the commands
  joint_position_tau: 0.3
  command_guard_variant: safe_stop # if requested position / velocities beyond limits, CommandGuard will be triggered and shut the connection. Available: [default, safe_stop]
  # exponential moving average time constant for external joint torque measurements [s]:
  #   Set tau > robot sample time for more smoothing on the external torque measurements
  #   Set tau << robot sample time for more smoothing on the external torque measurements
  external_torque_tau: 0.6
  # exponential moving average time constant for joint torque measurements [s]:
  #   Set tau > robot sample time for more smoothing on the raw torque measurements
  #   Set tau << robot sample time for more smoothing on the raw torque measurements
  measured_torque_tau: 0.6
  open_loop: true # KUKA works the best in open_loop control mode

estimated_ft_sensor: # estimates the external force-torque from the external joint torque values
  enabled: true # whether to enable the force-torque estimation
  update_rate: 100 # update rate for the force-torque estimation [Hz] (less or equal to controller manager update rate)
  rt_prio: 30 # real-time priority for the force-torque estimation
  chain_root: lbr_link_0
  chain_tip: lbr_link_ee
  damping: 0.2 # damping factor for the pseudo-inverse of the Jacobian
  force_x_th: 1.0 # x-force threshold. Only if the force exceeds this value, the force will be considered
  force_y_th: 1.0 # y-force threshold. Only if the force exceeds this value, the force will be considered
  force_z_th: 1.0 # z-force threshold. Only if the force exceeds this value, the force will be considered
  torque_x_th: 0.2 # x-torque threshold. Only if the torque exceeds this value, the torque will be considered
  torque_y_th: 0.2 # y-torque threshold. Only if the torque exceeds this value, the torque will be considered
  torque_z_th: 0.2 # z-torque threshold. Only if the torque exceeds this value, the torque will be considered
```

Please note

```yaml
  joint_position_tau: 0.3
  external_torque_tau: 0.6
  measured_torque_tau: 0.6
```

With high smoothing values.

As for the non-updating measured joint position, may I ask what version of this repository you are using? I think this was an issue but shouldn't be an issue anymore. Thank you!

mhubii, 2025-05-31:
okay, so you should be able to just pull the recent changes and this should fix it. I think this was fixed here: https://github.com/lbr-stack/lbr_fri_ros2_stack/pull/237

BoWangFromMars, 2025-05-31:
[heart]         Bo Wang reacted to your message:

closed 2025-05-31 by mhubii