lbr-stack/lbr_fri_ros2_stack #281: Where can I adjust PID parameters for joint position controller and velocity controller ?

Taokt, 2025-06-06:
I saw that there used to be a place to set up parameters for PID based on [this pose](https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/211#issuecomment-2422405830), but it is no longer there on the latest humble branch, it seems? 

Our Kuka robot always has a sudden jump at the start of every execution (after MoveIt plans the path), it happens for both position control and twist control, and I am trying to figure out if it could be caused by the PID controller not being set correctly.

Thanks in advance!

mhubii, 2025-07-04:
> * Are there some gain parameters that I can modify to adjust the force output of the controller, instead of manually setup the force?

Yes, you can modify the stiffness of the Cartesian Impedance controller in the Java application. A smaller stiffness `K` (`F = K*dx`) results in a larger `dx` for the same target force `F`.

Alternatively, @Hydran00 has implemented a Cartesian Impedance controller, which may be of help: https://github.com/idra-lab/kuka_impedance.

> * My goal is to eventually call the LBR Wrench Command Controller by passing the ee's target_pose, can the current wrench controller do this? If so, are there any instructions on how to achieve this?

This controller accepts joint positions and a target force. You will currently have to do inverse kinematics yourself.

Taokt, 2025-07-04:
Hi @mhubii , thank you for reply and clear things out, I am actually already trying the impedance controller that @Hydran00 wrote, but thank you for pointing the way!

Hi @Hydran00 , thank you for the extra information, I just tried the controller from [kuka-prop-ctrl](https://github.com/lucabeber/ros2_effort_controller/tree/kuka-prop-ctrl), I noticed that the name of the controller is different from the main branch, but I still have problem launching the controller after changing the name, but I guess we will continue the conversion under the issue of this (kuka impedance controller repo)[https://github.com/idra-lab/kuka_impedance].

And thank you both very much for answers!  It really helped me a lot!

I will close this issue.

closed 2025-07-04 by Taokt