lbr-stack/lbr_fri_ros2_stack #283: How to connect to FRI 1.17

dickson-wong, 2025-06-20:
Hi, I'd like to ask is there any suggestions on how I can connect to FRI 1.17

I saw the version 1.17 in fri, and I would like to use lbr_fri_ros_stack.

Is there any way I can do it?

Thank you so much for the help

dickson-wong, 2025-06-30:
Thanks for the reply.

I tried to follow the instruction. However, it stuck at the step when I input
“FRI_CLIENT_VERSION=1.17”

Thanks

dickson-wong, 2025-06-30:
I am following the instruction here:
https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_fri_ros2_stack/doc/lbr_fri_ros2_stack.html

In step 2, i did the following:
source /opt/ros/humble/setup.bash
export FRI_CLIENT_VERSION=1.17
mkdir -p lbr-stack/src && cd lbr-stack

After that I run this line:
"vcs import src --input https://raw.githubusercontent.com/lbr-stack/lbr_fri_ros2_stack/humble/lbr_fri_ros2_stack/repos-fri-${FRI_CLIENT_VERSION}.yaml"

An error message occur as follows:
HTTP Error 404: Not Found

dickson-wong, 2025-07-02:
Thanks for the reply.

When I check the repo, there is already one "fri-1.17" in the branch. So it suppose to be ok when I request FRI_CLIENT_VERSION=1.17

Should I still need to create a new branch named "fri-1.17?

Thanks

mhubii, 2025-07-02:
Yes the branch exists but the FRI is not 1.17.

You will have to
- Fork the repository
- Clone your fork locally
- Checkout the fri-1.17 branch
- Replace the zip file with your FRI
- Add, commit, push
- Open a pull request

mhubii, 2025-12-05:
Hi @SergeAlhalbi, 1.17 should already be available! Refer https://github.com/lbr-stack/fri.

The readme probably needs an update.

That said, I'd recommend working on Ubuntu 24, ROS 2 Jazzy.

SergeAlhalbi, 2025-12-05:
Wow, that was the quickest response I’ve ever received. I see, the reason I asked is that I noticed the open issue Server for 1.17 (https://github.com/lbr-stack/fri/issues/37), which mentions that the branch doesn’t actually host the correct version yet.

Thanks for the suggestion. I was trying to get several robot brands working on one Ubuntu setup as a first step. I might end up switching to the most supported version for each one separately.

dickson-wong, 2025-12-05:
Hi @SergeAlhalbi, the fri-1.17 branch should be working. 

By following the instruction in
https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_fri_ros2_stack/doc/hardware_setup.html

At least I can successfully finish the installation up to here:

[image]

However, I was not able to install the "LBRServer application".
As I said in the previous post, there suppose to have a file named "LBRServer.java" in the "server_app" folder.
It seems that most branches miss this folder and file.

 @mhubii, is there any way to solve this problem?

dickson-wong, 2025-12-05:
Hi @mhubii, thanks for that. Let me try to implement it next week.

May I ask where this file is from?
Is it in the KUKA FRI installation files? If so, how can I find it?

SergeAlhalbi, 2026-01-07:
Hi @dickson-wong and @mhubii, since I know you were trying to get this to work: before the holidays, I tested the FRI 1.15 server app from https://github.com/lbr-stack/fri/tree/ros2-fri-1.15/server_app with my FRI 1.17 setup, and it works. I was able to control a real KUKA LBR iiwa 7 robot through ROS2 and MoveIt.

dickson-wong, 2026-01-07:
Hi @SergeAlhalbi, thanks for the information.

Actually I also tried to use the same file, and it worked on my side also.