lbr-stack/lbr_fri_ros2_stack #301: Using FRI for the KMR IIWA

protoCore160, 2025-09-03:
Hello,

I have already used FRI alone on an LBR iiwa.
I now want to check if it is possible to use FRI on a KMR iiwa (KMP base + LBR).
The ROS 2 package found (kmr_iiwa - https://github.com/MortenMDahl/kmriiwa_ws_devel) should be fine for the base, but its arm control does not fit our needs. For context, see the diagram attached below (original communication layout) or the github link provided.

Note: We currently do not have physical access to the KMR and our time window is limited, so an early feasibility check and any blockers/gotchas would be very helpful.

Is it possible to run a single Sunrise application where:

- the base keeps its existing ROS 2 TCP interface (control + odometry/scanners), and

- the arm is controlled via FRI (UDP) using lbr_fri_ros2_stack?

Questions:

- Can FRI (UDP) run in parallel with the base TCP communication within the same Sunrise application?

- Is this setup feasible?

Thanks!

[image]

mhubii, 2025-09-05:
Hi @protoCore160, sorry I have no experience. 

Generally speaking, the FRI on Sunrise runs blocking. So to run another communication, one would have to run it asynchronously, which seems possible in Java. 

But then again, I've never gotten to use a base.

LAKY911, 2025-11-28:
@protoCore160 Great, how did you combine both the lbr_fri_ros2_stack with the KMR? Are you willing to share the code which you've uploaded to the controller?

Thank you!