lbr-stack/lbr_fri_ros2_stack #302: Can't Connect To KUKA iiwa 14. Any way to check installed FRI Version without Sunrise Workbench (directly from teach pedant)?

ian-chuang, 2025-09-25:
Hello,

I have a used KUKA iiwa 14 that was previously owned from another research lab. Due to circumstances, there used to be some properly working code using lbr_fri_ros2_stack that controlled the robot but it was deleted. As a result, I both don't have access to the software (Sunrise Workbench, FRI), but I know that some version of FRI was installed on the robot that had previously worked.

I've tried 
```
ros2 launch lbr_bringup hardware.launch.py     model:=iiwa14 
```
using all the available FRI versions (1.11, 1.14, 1.15, 1.16, 2.5, 2.7) but get the same issue here:
```
[ros2_control_node-2] [INFO] [1758833628.824701436] [lbr_ros2_control::SystemInterface]: Awaiting robot heartbeat. Attempt 62, remote_host 'INADDR_ANY', port_id '30200'
```

Is it possible to check the FRI version installed through the teach pendant? Or do you know if I provide serial number, etc to KUKA they will provide necessary software to properly install FRI again?

mhubii, 2025-09-26:
thanks for sharing @ian-chuang. Sorry, missing some background here. 

Quick reminder: FRI communicates client (your laptop / computer running this stack) <-> server (KUKA controller running FRI server application)

Two immediate questions:

- Did you configure your IP correctly on your client side (laptop / computer)?
- Were you able to `ping 172.31.1.147` ? (https://lbr-stack.readthedocs.io/en/latest/lbr_fri_ros2_stack/lbr_fri_ros2_stack/doc/hardware_setup.html#verify-connection-to-the-robot). The IP may vary depending on `X66` / `KONI` ports
- Did you run an FRI server application on the KUKA controller via the pendant?

mhubii, 2025-09-26:
you should be able to get it to run with FRI `1.10`

ian-chuang, 2025-09-26:
Hi,

So I'm trying to get the branch fri-1.11 to work with my robot which has installed FRI 1.10.

So i reinstalled FRI 1.10 and LBRServer to the robot with Sunrise Workbench. I noticed that on the fri-1.11 branch https://github.com/lbr-stack/fri/tree/fri-1.11 there is no `serverapp` folder so I used the LBRServer.java from https://github.com/lbr-stack/fri/tree/ros2-fri-1.15 . Is that okay?

Next I do the following:
- Setup robostack ros2 humble environment
- Setup workspace
```
export FRI_CLIENT_VERSION=1.11
mkdir -p lbr-stack/src && cd lbr-stack
vcs import src --input https://raw.githubusercontent.com/lbr-stack/lbr_fri_ros2_stack/humble/lbr_fri_ros2_stack/repos-fri-${FRI_CLIENT_VERSION}.yaml
rosdep install --from-paths src -i -r -y
```
- I go into src/fri and replace FRI-Client-SDK_cpp.zip with the file I got from Sunrise Workbench, then update the version check to 1.10
- Next i do `colcon build`, however I don't do `colcon build --symlink-install` since I get the error below. Wondering if just doing colcon build is fine?
```
(ros_env) ianchuang@computer:~/lbr-stack$ colcon build --symlink-install
Starting >>> lbr_description
Starting >>> fri_client_sdk
Starting >>> lbr_moveit
Starting >>> lbr_moveit_cpp
/home/ianchuang/miniforge3/envs/ros_env/lib/python3.11/site-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
Finished <<< fri_client_sdk [0.19s]
Starting >>> lbr_fri_idl
--- stderr: lbr_moveit
/home/ianchuang/miniforge3/envs/ros_env/lib/python3.11/site-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --editable not recognized
---
Failed   <<< lbr_moveit [0.56s, exited with code 1]
Aborted  <<< lbr_description [0.60s]
Aborted  <<< lbr_moveit_cpp [1.63s]
Aborted  <<< lbr_fri_idl [1.74s]
```
- After `colcon build` and `source install/setup.bash` I run:
```
ros2 launch lbr_bringup hardware.launch.py  model:=iiwa14
```
- Then I run LBR Server on the pendant (10ms, 172.31.1.148, position_control, position)
- And stuck at:
```
[ros2_control_node-2] [INFO] [1758914353.923230591] [lbr_ros2_control::SystemInterface]: Awaiting robot heartbeat. Attempt 16, remote_host 'INADDR_ANY', port_id '30200'
```

I can ping the robot
```
(base) ianchuang@computer:~/lbr-stack$ ping 172.31.1.147
PING 172.31.1.147 (172.31.1.147) 56(84) bytes of data.
64 bytes from 172.31.1.147: icmp_seq=1 ttl=64 time=0.388 ms
```

I also can see that when I run LBRServer on the pendant I can listen in on data it is sending:
```
(base) ianchuang@computer:~/lbr-stack$ sudo tcpdump -i enp36s0f1 port 30200
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on enp36s0f1, link-type EN10MB (Ethernet), snapshot length 262144 bytes
19:22:06.758902 IP 172.31.1.147.30200 > hesse.30200: UDP, length 386
19:22:06.768812 IP 172.31.1.147.30200 > hesse.30200: UDP, length 386
19:22:06.778533 IP 172.31.1.147.30200 > hesse.30200: UDP, length 386
19:22:06.788495 IP 172.31.1.147.30200 > hesse.30200: UDP, length 386
```

running `ifconfig`
```
enp36s0f1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.31.1.148  netmask 255.255.255.0  broadcast 172.31.1.255
```

On ubuntu network settings I just set static IP to 172.31.1.148 and netmask 255.255.255.0

Thanks

ian-chuang, 2025-09-26:
On the teach pendant after running for 10 seconds the LBRServer app gives the same error as in https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/68#issue-1631057912

Except it says creating FRI connection to 172.31.1.148 instead

ian-chuang, 2025-09-26:
Hi @mhubii,

Sorry to ping so much, just forgot to add @ here's the error message on pendant

[image]

ian-chuang, 2025-09-29:
Hello,

No problem, thanks a lot for the help already. I tried the standalone applications. Replaced `FRI-Client-SDK_Cpp.zip` with version 1.10 from Sunrise Workbench. And modified CMakeLists.txt version to 1.10. Here is output after running:
```
(ros_env) ianchuang@computer:~/GitHub/fri$ ./build/LBRJointSineOverlayApp
LBRJointSineOverlayClient initialized:
        joint mask: 0x8
        frequency (Hz): 0.250000
        amplitude (rad): 0.040000
        filterCoeff: 0.990000
```
And while that is running I also run LBRJointSineOverlay on pendant and get this error
[image]

I checked the connection and can ping the robot still
```
(ros_env) ianchuang@computer:~/GitHub/fri$ ping 172.31.1.147
PING 172.31.1.147 (172.31.1.147) 56(84) bytes of data.
64 bytes from 172.31.1.147: icmp_seq=1 ttl=64 time=0.554 ms
```

mhubii, 2025-09-30:
Sure, try that. But it is confusing that no connection is established at all.

Maybe the CMake support misses some files for 1.10? Have you tried just compiling through KUKA's make files?

ian-chuang, 2025-10-03:
Hello I tried the make files that came with KUKA FRI Client SDK but get the same result where connection times out. You are referring to makefiles in build/GNUMake?
```
~/Downloads/FRI-Client-SDK_Cpp/build/GNUMake$ make

...

~/Downloads/FRI-Client-SDK_Cpp/build/GNUMake$ ../../bin/LBRJointSineOverlay 
LBRJointSineOverlayClient initialized:
        joint mask: 0x8
        frequency (Hz): 0.250000
        amplitude (rad): 0.040000
        filterCoeff: 0.990000
```

I can see some data going from robot to computer when I run LBRJointSineOverlay on the robot pendant:
```
(base) ianchuang@hesse:~/lbr-stack/src/fri$ sudo tcpdump -i enp36s0f1 port 30200
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on enp36s0f1, link-type EN10MB (Ethernet), snapshot length 262144 bytes
20:02:36.055579 IP 172.31.1.147.30200 > 172.31.1.148.30200: UDP, length 386
20:02:36.060576 IP 172.31.1.147.30200 > 172.31.1.148.30200: UDP, length 386
20:02:36.065557 IP 172.31.1.147.30200 > 172.31.1.148.30200: UDP, length 386
```

I believe I have a "LBR iiwa 14 R820 RAL 9016" as listed on my.KUKA so I don't think it is the Medical KUKA. Do you know how to do the brake test? I only see this on the manual about a "brake test"
```
Test of the functional capability of the brakes:
For the industrial robot, a brake test is available which can be used to
check whether the brake of each axis applies sufficient braking torque.
The brake test ensures that any impairment of the braking function is de-
tected, e.g. due to wear, overheating, fouling or damage, thereby eliminat-
ing avoidable risks.
The brake test must be performed regularly, unless an application-specific
risk assessment has established that a malfunction of the mechanical
brakes will not result in inadmissibly high risks. Determination of the inter-
val at which the brake test is to be performed also constitutes part of the
risk assessment.
In the absence of a corresponding risk assessment, the following applies:
• The brake test must be carried out for each axis during start-up and
recommissioning of the industrial robot.
• The brake test must be performed daily during operation.
```

ian-chuang, 2025-10-03:
@mhubii 

Also, I forgot to mention this but do you know the cause of this warning listed in `Safety/Status`? I've always had this warning and wondering if its messing with the connection.

[image]

ian-chuang, 2025-10-03:
Ah nevermind, I resolved the warnings by removing some of the safety configs and uploading and running PositionAndGMSReferencing.java . I also have run brake test with BrakeTestApplication.java . However still can not form any connection with either LBRServer.java (lbr_fri_ros2) or LBRJointSineOverlay.java (FRI Client SDK)

ian-chuang, 2025-10-03:
Finally fixed the issue !!!

It was an issue with the firewall. Even though I could view data with `sudo tcpdump -i enp36s0f1 port 30200` for whatever reason trying to read udp data with either C++ or python script didn't work until I did this:

```
sudo ufw allow 30200/udp
sudo ufw reload
```

Thanks so much again

closed 2025-10-03 by ian-chuang

mhubii, 2025-10-04:
> do you know the cause of this warning listed

The position reference warnings can be resolved by moving the joints +/- 10 degrees. That way, the Hall sensors will reference the zero position.

> Is a real-time capable computer with real-time capable operating system necessary? I don't think I've enabled realtime capability...

Not necessarily.

> It was an issue with the firewall.

Fantastic :), good find! So was probably as in https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/67#issuecomment-1479794623. I should have caught that!

Thanks for the quick update and the detailed feedback all along.